OpenKinect/libfreenect2 #618: Link Error: libfreenect2::Freenect2Device::Config config;

ronin8600, 2016-03-16:
I'm trying to enable EnableBilateralFilter by doing the following

libfreenect2::Freenect2Device::Config config;
config.EnableBilateralFilter = true;
device->setConfiguration(config);

I get a link error trying to find libfreenect2::Freenect2Device::Config::Config(void)

Is this a known issue? 

Version, Platform, and Hardware Bug Found:

1.  803e7fb Merge pull request #614 from hanyazou/status_1024
2.  Windows 7 - Visual Studio 12 (2013)

ronin8600, 2016-03-16:
It looks like the Config constructor is not getting exported.

In libfreenect2.hpp I changed "struct Config" to "struct LIBFREENECT2_API Config" and it now compiles.

xlz, 2016-03-17:
Add here
https://github.com/OpenKinect/libfreenect2/blob/master/include/libfreenect2/libfreenect2.hpp#L192

```
class LIBFREENECT2_API Freenect2Device::Config;
```
Does this work?

ronin8600, 2016-03-18:
xlz,

Added your suggested fix and it doesn't compile.

Is there an objection with my proposed fix?

https://github.com/OpenKinect/libfreenect2/blob/master/include/libfreenect2/libfreenect2.hpp#L115
Changing "struct Config" to "struct LIBFREENECT2_API Config"

xlz, 2016-03-18:
No objection, just saying.

I created this fix https://github.com/OpenKinect/libfreenect2/commit/a650bc7e628e589f322df8c7ee1384ba131497a4.

closed 2016-03-18 by xlz