OpenKinect/libfreenect2 #619: OpenNI2: convertDepthToWorld not working properly

rafapages, 2016-03-18:
Hi everyone!

I'm having an issue when transforming from depth values to world coordinates using OpenNI. Although it may seem as it worked fine, the resulting point cloud is deformed. I think this is because the depth frames are set to 640x480 to allow skeleton tracking through NiTE2, and this is messing up the transformation. 

I'm pretty sure that is the problem because if you force the depth stream to 512x424, the transformation is correct and the point cloud is perfectly fine (but, of course, no skeleton is tracked). But I need the skeleton too, so I cannot simply do this.

OpenNI is probably using the centre of the frame or its dimensions automatically for the intrinsic parameter matrix, and it's not taking into account the original information...

Is it possible to fix this?

Thanks a lot!

hanyazou, 2016-03-18:
@rafapages sorry, i have no idea.

>transforming from depth values to world coordinates using OpenNI.

Could you tell me how you transform them exactly? Did you use some values from OpenNI adding to the depth values? Can we fix this problem if we change the values?

rafapages, 2016-03-18:
I'm using the function convertDepthToWorld. Something like this:
```
openni::CoordinateConverter coorConverter;
float worldX, worldY, worldZ = 0.0;
for (int y = 0; y < depthHeight_ ; y++){
  for (int x = 0; x < depthWidth_; x++){
    const int index = y * depthWidth_ + x;
    coorConverter.convertDepthToWorld (depthStream_, x, y, depthImage[index], &worldX, &worldY, &worldZ);
    realWorld[index]=Vec3f(worldX,worldY,worldZ);
  }
}
```
Where `depthImage` contains the depth values for the whole frame and `depthStrem_` is a `openni::VideoStream`. If `depthStream_` is initialized with 512x424, the transformation is correct, but the skeleton cannot be tracked. If `depthStream_` is initialized with 640x480, the transformation is wrong, but the skeleton is tracked perfectly. 

I'm going to search in the code. The problem must be in the coverDepthToWorld function, I guess.

hanyazou, 2016-03-18:
@rafapages I'd like to investigate this problem if you provide whole, compilable source code.

rafapages, 2016-03-18:
I'd like to, but this is just a tiny part of a huge project. I'm probably not even allowed to share it all...

I'm searching for the implementation of the convertDepthToWorld function in the openni2 directory, but I cannot find it... Could this be the problem? I might just be using the default openni conversion, right?