OpenKinect/libfreenect2 #620: Error during installation on OSX (missing:  PKG_CONFIG_EXECUTABLE)

MrfksIv, 2016-03-18:
After installing all the packages needed, i created the build folder and cd-ed there. But when i run the command `cmake ..` i get the following error:

`-- using tinythread as threading library
-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
CMake Error at /sw/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find LibUSB (missing: LibUSB_LIBRARIES)
Call Stack (most recent call first):
  /sw/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  cmake_modules/FindLibUSB.cmake:87 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:93 (FIND_PACKAGE)`

After a couple of hours of searching online i am still not closer to solving this. Can anyone help?

Thanks

hanyazou, 2016-03-18:
Did you install pkg config package?
$ brew install pkgconfig

xlz, 2016-03-18:
> Make sure these build tools are available: wget, git, cmake, pkg-config. Xcode may provide some of them. Install the rest via package managers.

closed 2016-03-18 by xlz

v2k, 2016-04-14:
@hanyazou @xlz 
`brew install pkgconfig `

This fixed my issue too, can we add this to the OSX install readme?

xlz, 2016-04-14:
From README:

> Make sure these build tools are available: wget, git, cmake, **pkg-config**. Xcode may provide some of them. Install the rest via package managers.

nijisakai, 2024-11-28:
> brew install pkgconfig

like a charm