OpenKinect/libfreenect2 #624: Problem with two kinect v2 devices

MM88, 2016-04-01:
Overview Description:

Hi, I'm trying to run two devices at the same time, I have to following code:

` void run () {`
`       //first kinect`
  `       std::string device_id1 = "#"+boost::lexical_cast<std::string>(1);`
    `     pcl::io::OpenNI2Grabber interface1 (device_id1);`
      `   boost::function<void (const PointCloud<PointXYZRGBA>::ConstPtr&)> f = boost::bind(&SimpleOpenNIViewer::grabberCallback,this, _1);`
        ` interface1.registerCallback (f);`
        ` interface1.start ();`
      ` //second kinect`
       `  std::string device_id2 = "#"+boost::lexical_cast<std::string>(2);`
        ` pcl::io::OpenNI2Grabber interface2 (device_id2);`
       `  boost::function<void (const PointCloud<PointXYZRGBA>::ConstPtr&)> f2 = boost::bind(&SimpleOpenNIViewer::cloud_cb_2_,this, _1);`
       `  interface2.registerCallback (f2);`
        ` interface2.start ();`
       `  while (!viewer.wasStopped())`
       `  {  sleep (1);  }`
       `  interface2.stop ();`
        ` interface1.stop ();`
    `}`

This code works fine with two PrimeSense devices and with one Kinect2 and one PrimeSense, but with two kinect2 it open two times the same device.
It seams that when I have two kinect2 connected I can't adress them, nor with device id or serial number.

Thanks in advance.

xlz, 2016-04-01:
The code is using PCL interfaces. You should report to PCL not here.

closed 2016-04-01 by xlz

hanyazou, 2016-04-07:
@MM88 This might be a bug of libfreenect2 OpenNI2 driver. Could you provide whole source code which can be compiled and run? I must investigate this problem.

MM88, 2016-04-07:
Hi, I found out that the second device needed to be addressed with "#5" not "#2", sorry to bother

smokhov, 2016-04-07:
Maybe related to #603.