OpenKinect/libfreenect2 #625: running libfreenect2 and kinect2 without X

mpatalberta, 2016-04-02:
The current configuration of the Kinectv2 usb configuration requires that the user has a graphical display.
If you want to use the kinectv2 without the gui (X) for example. The following usb rule could be used:Ensure you can access the sensor
Give the sensor power and plug the usb connector from the sensor to your linux box. Verify you have the correct udev rules file to be able to access the sensor. You can verify by looking for the file /etc/udev/rules.d/70-kinect2.rules. It should contain the following:

BUS!="usb", SUBSYSTEM!=="usb_device", ACTION!="add", GOTO="kcontrol_rules_end"
# this file belongs in /etc/udev/rules.d/
# ATTR{product}=="Kinect2"
SUBSYSTEM=="usb", ATTRS{idVendor}==“045e", ATTRS{idProduct}=="02c4", MODE:="0666", OWNER:="root", GROUP:="video"
SUBSYSTEM=="usb", ATTRS{idVendor}==“045e", ATTRS{idProduct}=="02d8", MODE:="0666", OWNER:="root", GROUP:="video"
SUBSYSTEM=="usb", ATTRS{idVendor}==“045e", ATTRS{idProduct}=="02d9", MODE:="0666", OWNER:="root", GROUP:="video"

LABEL="kcontrol_rules_end"

If it doesn't exist, create it using sudo access and fill the contents with the text shown above.

The rules allow you to access the device if you are a member of the video group (or you are root of course). Make sure you are a member of the video group.

arwillis@visionlab16-pc:~/NRC/ROS_RGBD_Odometry/ros_ws/src$ cat /etc/group | grep video video:x:44:arwillis
This shows my username arwillis next to the output line starting with video in the /etc/groups file. If you don't see your name next to the video group as above you must add yourself to the video group with the following command sudo adduser <myusername> video. Re-run the command above to verify you are now a member of the video group. If you are newly added to the video group, YOU MUST LOGOUT AND LOG BACK IN TO MAKE YOUR NEW GROUP ASSOCIATION/PERMISSIONS ACTIVE.

de-vri-es, 2016-04-02:
These instructions are specific to whichever Linux distro you are using. For example, on Arch Linux, the use of TAG+="uaccess" is the preferred method of granting permissions on a device to the active seat. The burden of adding the proper udev rules lies with packagers for a specific distro, although the example included is very useful for getting the proper matching rules.

mpatalberta, 2016-04-02:
Maarten,
There is something I did not know. I thought most of the Linux systems used the same style udev setup.

Pat

________________________________
- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

de-vri-es, 2016-04-02:
It's true that pretty much all Linux systems use udev, but there is a lot of variety in the usage of groups, extended acls, systemd integration, etc. That makes writing universal udev rules pretty much impossible.

mpatalberta, 2016-04-02:
The install applied to Ubuntu 14.04 desktop.

xlz, 2016-04-11:
I think this is covered in README.

closed 2016-04-11 by xlz