OpenKinect/libfreenect2 #626: Can't run multiple Protonects

MohitShridhar, 2016-04-03:
Can't run `./Protonect 004566752747` and then another `./Protonect 004674752747`. The second Protonect fails to startup.

Debug Log:
[https://gist.github.com/MohitShridhar/428c4c800f89c39889ca8a774bc4a61e](https://gist.github.com/MohitShridhar/428c4c800f89c39889ca8a774bc4a61e)

I have tried increasing the USBFS buffer size to `64` and `128` with no luck. I am trying to get two Kinects to work on a MacbookPro 2013 (16GB RAM) running Ubuntu 14.04 (64bit) with each sensor connected to a separate bus.

I tried setting the large transfer environment variables specified in the Troubleshooting Guide, but when I set all four, even a single Protonect fails to startup.

`lsusb -t`:
> /:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
>     |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/1p, 5000M
>         |__ Port 1: Dev 9, If 0, Class=Vendor Specific Class, Driver=, 5000M
>         |__ Port 1: Dev 9, If 1, Class=Vendor Specific Class, Driver=, 5000M
>         |__ Port 1: Dev 9, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
>         |__ Port 1: Dev 9, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
>     |__ Port 2: Dev 5, If 0, Class=Hub, Driver=hub/1p, 5000M
>     |__ Port 4: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
> /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
>     |__ Port 1: Dev 7, If 0, Class=Hub, Driver=hub/1p, 480M
>     |__ Port 8: Dev 2, If 0, Class=Hub, Driver=hub/3p, 12M
>         |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=btusb, 12M
>         |__ Port 3: Dev 6, If 1, Class=Wireless, Driver=btusb, 12M
>         |__ Port 3: Dev 6, If 2, Class=Vendor Specific Class, Driver=, 12M
>         |__ Port 3: Dev 6, If 3, Class=Application Specific Interface, Driver=, 12M
>     |__ Port 12: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
>     |__ Port 12: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
>     |__ Port 12: Dev 3, If 2, Class=Human Interface Device, Driver=bcm5974, 12M

`lspci -nn`:

> 00:00.0 Host bridge [0600]: Intel Corporation Crystal Well DRAM Controller [8086:0d04] (rev 08)
> 00:01.0 PCI bridge [0604]: Intel Corporation Crystal Well PCI Express x16 Controller [8086:0d01] (rev 08)
> 00:01.1 PCI bridge [0604]: Intel Corporation Crystal Well PCI Express x8 Controller [8086:0d05] (rev 08)
> 00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 05)
> 00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
> 00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 05)
> 00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d5)
> 00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 [8086:8c14] (rev d5)
> 00:1c.3 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #4 [8086:8c16] (rev d5)
> 00:1c.4 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 [8086:8c18] (rev d5)
> 00:1f.0 ISA bridge [0601]: Intel Corporation HM87 Express LPC Controller [8086:8c4b] (rev 05)
> 00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 05)
> 01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GK107M [GeForce GT 750M Mac Edition] [10de:0fe9] (rev a1)
> 01:00.1 Audio device [0403]: NVIDIA Corporation GK107 HDMI Audio Controller [10de:0e1b] (rev a1)
> 03:00.0 Network controller [0280]: Broadcom Corporation BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)
> 04:00.0 Multimedia controller [0480]: Broadcom Corporation Device [14e4:1570]
> 05:00.0 SATA controller [0106]: Samsung Electronics Co Ltd Apple PCIe SSD [144d:1600] (rev 01)

xlz, 2016-04-04:
A Not enough bandwidth problem.

MohitShridhar, 2016-04-05:
Thanks! My Macbook Pro has two USB3 ports with two separate buses. Technically bandwidth shouldn't be a problem right? Could it be some driver issues?

closed 2016-04-08 by MohitShridhar