OpenKinect/libfreenect2 #627: Cannot find `libusb` which I have installed

zry656565, 2016-04-07:
Overview Description:

I followed https://github.com/OpenKinect/libfreenect2/blob/master/README.md#linux to install libfreenect2, and the log of cmake told me that it couldn't find `libusb` which I had installed.

```
jerry@jworktu:~/lab/libfreenect2/build$ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'libusb-1.0>=1.0.20'
--   package 'libusb-1.0>=1.0.20' not found
CMake Error at /usr/share/cmake-3.2/Modules/FindPkgConfig.cmake:344 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.2/Modules/FindPkgConfig.cmake:506 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
  CMakeLists.txt:93 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/home/jerry/lab/libfreenect2/build/CMakeFiles/CMakeOutput.log".

jerry@jworktu:~/lab/libfreenect2/build$ sudo apt-get install libusb-1.0-0-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 180 not upgraded.
```

**Version, Platform, and Hardware Bug Found:**

**`git log -1 --oneline`**
```
619bd52 depends: Do not download libva debs for non-x86
```

**`uname -a`**
```
Linux jworktu 4.2.0-35-generic #40-Ubuntu SMP Tue Mar 15 22:15:45 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

**`lsusb -t`**
```
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 1: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 1: Dev 8, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 2: Dev 6, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 3: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 5, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 5: Dev 5, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 7: Dev 4, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 4, If 1, Class=Video, Driver=uvcvideo, 480M
```
**`lspci -nn`**

```
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor DRAM Controller [8086:0c04] (rev 06)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller [8086:0c01] (rev 06)
00:02.0 VGA compatible controller [0300]: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller [8086:0416] (rev 06)
00:03.0 Audio device [0403]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller [8086:0c0c] (rev 06)
00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 05)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 05)
00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 05)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d5)
00:1c.1 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 [8086:8c12] (rev d5)
00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 [8086:8c14] (rev d5)
00:1c.3 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #4 [8086:8c16] (rev d5)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 05)
00:1f.0 ISA bridge [0601]: Intel Corporation HM86 Express LPC Controller [8086:8c49] (rev 05)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c03] (rev 05)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 05)
01:00.0 3D controller [0302]: NVIDIA Corporation GM107M [GeForce GTX 860M] [10de:1392] (rev a2)
04:00.0 Network controller [0280]: Qualcomm Atheros AR9485 Wireless Network Adapter [168c:0032] (rev 01)
05:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. Device [10ec:5287] (rev 01)
05:00.1 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 12)
```

floe, 2016-04-07:
Thanks for the thorough report. What does `dpkg -l libusb*` show?

zry656565, 2016-04-10:
Sorry for late reply, here it is.

```
jerry@jworktu:~/lab/Kintinuous/build$ dpkg -l libusb*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libusb-0.1-4:a 2:0.1.12-27  amd64        userspace USB programming library
ii  libusb-1.0-0:a 2:1.0.19-1   amd64        userspace USB programming library
ii  libusb-1.0-0-d 2:1.0.19-1   amd64        userspace USB programming library
ii  libusb-1.0-doc 2:1.0.19-1   all          documentation for userspace USB p
un  libusb-dev     <none>       <none>       (no description available)
un  libusbmuxd-too <none>       <none>       (no description available)
ii  libusbmuxd2:am 1.0.9-1build amd64        USB multiplexor daemon for iPhone
```

floe, 2016-04-10:
OK, so you still have libusb-1.0.19 installed. IIRC our PPA doesn't support 15.10, so as a workaround, I'd suggest you download the .deb files from https://launchpad.net/ubuntu/xenial/+source/libusb-1.0 and install those manually; shouldn't be a problem IMHO. In any case, I'd recommend upgrading to 16.04 once it's out.

xlz, 2016-04-11:
That's pretty much it. `package 'libusb-1.0>=1.0.20' not found` because you only had libusb-1.0 1.0.19.

closed 2016-04-11 by xlz

CodeFreaklLinux, 2017-11-01:
Dear friends,
Hope everything be at your convenience.
I'm beginner in programing. I'v installed libusb-1.0 and dpkg -l libusb* shows;
```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                     Version           Architecture      Description
+++-========================-=================-=================-=====================================================
ii  libusb-0.1-4:amd64       2:0.1.12-28       amd64             userspace USB programming library
ii  libusb-1.0-0:amd64       2:1.0.20-1        amd64             userspace USB programming library
ii  libusb-1.0-0-dbg:amd64   2:1.0.20-1        amd64             userspace USB programming library development files
ii  libusb-1.0-0-dev:amd64   2:1.0.20-1        amd64             userspace USB programming library development files
ii  libusb-1.0-doc           2:1.0.20-1        all               documentation for userspace USB programming
un  libusb-dev               <none>            <none>            (no description available)
un  libusbmuxd-tools         <none>            <none>            (no description available)
ii  libusbmuxd4:amd64        1.0.10-2ubuntu0.1 amd64             USB multiplexor daemon for iPhone and iPod Touch devi
```
but after debugging my program following note, appears;
```
(gdb) s
libusb_init (context=0x0) at ../../libusb/core.c:1983
1983	../../libusb/core.c: No such file or directory.
(gdb) quit
```
Please let me know what should I do.
Thanks

floe, 2017-11-01:
@CodeFreaklLinux this is unrelated to the original issue. It just means that you don't have the source code for libusb-1.0 installed so you can't view the exact source line while debugging. Note that this only applies to the internal code of libusb, not to your program itself.