OpenKinect/libfreenect2 #628: [Kinect2Bridge::initDevice] could not stop device!

caselitz, 2016-04-08:
I am using the iai kinect2_bridge and there seems to be an issue with newer libfreenect2 versions, see: https://github.com/code-iai/iai_kinect2/issues/256

Any ideas what could be the problem?

xlz, 2016-04-09:
I checked. Every error path in stop() is covered with error messages. There is no error message before `[Kinect2Bridge::initDevice] could not stop device!`.

I don't know. See if you can reproduce this with Protonect. (Change `dev->stop()` to `if (!dev->stop()) std::cerr << "stop() failed" << std::endl;`)

caselitz, 2016-04-11:
I could not reproduce it with Protonect, which worked fine all the time.
Since I solved the issue by recompiling the iai ros package (*), I don't think the problem is in libfreenect2.

(*) Unfortunately, I'm also not able to reproduce the problem in the iai package (only if I use the old binaries). So no clue what caused the problem... :-/

xlz, 2016-04-11:
That's all right. Report it next time when you have it.

closed 2016-04-11 by xlz