OpenKinect/libfreenect2 #629: Need aligned color and depth frame

niteshtm, 2016-04-09:
Hi there!!

I am using libfreenect2 . Basically I need aligned Depth and RGB image. Using the registration class we can register the images but the registered image has black pixel where it does find depth data like at the edges or at a point very far.  How can I get aligned colored image without those black pixels?

floe, 2016-04-09:
That's a fundamental limitation: when there's no depth data, there's nothing to align to. What you can try is do the inverse mapping (depth onto color) by passing a `Frame` object to the optional `bigdepth` parameter in `Registration::apply` (see also https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html#a8be3a548ff440e30f5e89a4b053e04fa).

niteshtm, 2016-04-10:
Hi Floe !! I totally understand that it is a fundamental problem . By doing registration we are getting aligned image but since we generating color image based on depth data we can't get only colored image . Secondly if we are using bigdepth we won't be having depth data available for full rgb image as we can see there are full dark strips on sideways. 
As you can see in the first attached image i.e .image1, the difference between warped_registered_image and warped_rgb_image there is a shift between them so my alignment problem is solved but just wanted to know if by any means or any algorithm we can get this shift and get only colored image.
[image]
[image]
[image]

Thanks.

floe, 2016-04-10:
To be honest, I still don't quite understand the problem. The depth and color cameras have different fields of view, so you will always get some bars (either top/bottom or on the left side) where they don't overlap and where you consequently can't do any registration.

niteshtm, 2016-04-11:
Got it . How can i access the depth array from undistorted image ?

niteshtm, 2016-04-11:
Got everything worked . Thanks for all your help . I liked libfreenect2 . Just wondering if you guys are looking forward to have skeleton data , hands data or something like green screening in the library. But anyways I loved it. Thanks a lot guys. :)

closed 2016-04-11 by xlz

floe, 2016-04-11:
For skeleton data, you can try the OpenNI integration (see README).

rorpis, 2018-12-23:
Hi,
Currently I'm working on a project were I need to do image processing in python (using the rgb frame). How could you get the depth value of a pixel from the rgb frame, or map a rgb pixel to a depth pixel using the python wrapper?

buaazsj, 2019-08-19:
> Hi,
> Currently I'm working on a project were I need to do image processing in python (using the rgb frame). How could you get the depth value of a pixel from the rgb frame, or map a rgb pixel to a depth pixel using the python wrapper?

Hi,
I am doing the same thing, I want to map RGB pixels to depth data. Have you solved it?
I tried to align all the rgb pixels into the depth map, comment out the lines of code in a.cpp, and keep those backgrounds, but the actual effect is still only the foreground of the depth map, the background still has no data.
[image]