OpenKinect/libfreenect2 #632: Skeleton tracking using Kinect version 2 on ubuntu

bxs1260, 2016-04-12:
Overview Description: Thanks a ton for providing this useful tutorial. I did set-up everything successfully. I was wondering if someone has already worked with getting the skeleton tracking from Kinect version 2. If yes, please let me know how to achieve that.

My platform is: Linux linux 4.2.0-35-generic #40~14.04.1-Ubuntu  x86_64 x86_64 x86_64 GNU/Linux

Thanks and regards.

rafapages, 2016-04-12:
Yeah, I've been using it through the OpenNI2 framework and NiTE. Check out #243.

xlz, 2016-04-12:
I believe the above answers your question.

closed 2016-04-12 by xlz

bxs1260, 2016-04-13:
I did set-up libfreenect2, OpneNI2 and NiTE-Linux-x64-2.2 successfully but still I am unable to run : ~/NiTE-Linux-x64-2.2/Samples/Bin$ ./UserViewer 
Failed to open device
	DeviceOpen using default: no devices found

even ~/OpenNI2/Bin/x64-Release$ ./SimpleRead 
Couldn't open device
	DeviceOpen using default: no devices found

Thread mentions that depth resolution must be 640x480 and device name must be "Kinect" exactly. ("Kinect 2" does not work for the NiTE.) but I couldn't find where do to do this change.

Please help me in setting this up to get all the sample files running for skeleton tracking

xlz, 2016-04-13:
NiViewer2 is the command to test libfreenect2 with openni2. That's all I know.

bxs1260, 2016-04-13:
Thanks @xlz for quick reply. NiViewer2 and Protonect are working flawlessly from first attempt of setting this up. Now my main concern is to get the samples running from ~/NiTE-Linux-x64-2.2/Samples/Bin for skeleton tracking,currently Which I am unable to get.

~/OpenNI2/Bin/x64-Release$ NiViewer2
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 502335342542
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 502335342542
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [OpenGLDepthPacketProcessor] avg. time: 16.6058ms -> ~60.2197Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.6038ms -> ~53.7524Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 13.379ms -> ~74.7438Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.4286ms -> ~48.951Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.3064ms -> ~69.8988Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.7998ms -> ~53.1921Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.2372ms -> ~70.2385Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.6963ms -> ~53.4866Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.1058ms -> ~70.8928Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.7669ms -> ~53.2852Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.0873ms -> ~70.986Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.0334ms -> ~52.5392Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 18.556ms -> ~53.8908Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.6494ms -> ~53.6211Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 19.2594ms -> ~51.9227Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.1776ms -> ~52.1443Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 19.1305ms -> ~52.2724Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.2124ms -> ~52.0497H

bxs1260, 2016-04-13:
@rafapages need your guidance. Thanks

rafapages, 2016-04-19:
Hi @bxs1260,

I don't understand your problem. It looks that you are unable to open the Kinect, so it's not a NiTE problem. Are you sure UserViewer is running using the correct version of OpenNi and libfreenect?

bxs1260, 2016-04-21:
Hello @rafapages ,

Thanks for a quick reply. I am able to open Kinect using only these two: NiViewer2 and Protonect. I am not able to use UserViewer or any other program from ~/NiTE-Linux-x64-2.2/Samples/Bin . I really appreciate your help on this issue.

I have another concern too,Once I manage to get skeleton tracking using Kinect v2 then I want to simulate Kinect v2 in gazebo for faster development of my project. Unfortunately I couldn't find much about doing this. So I am forced to go back to Kinect v1 though it doesn't provide precise data when compared to Kinect v2. I have a working setup of Kinect v1 and its simulation in gazebo.

hanyazou, 2016-04-23:
@bxs1260 I guess that UserViewer uses it's own openni2 library (.so) which is different from NiViewer2's. Use ldd command to find the library location and try to install (copy or symlink) libfreenect2 driver to the NiTE's openni2 location. It works on my Mac.

LovelyHorse, 2016-05-07:
Hey @bxs1260, I'm having the exact same issue you describe. 

    ldd UserViewer | grep ni 

Isn't very helpful though NiViewer2 and Protonect work perfectly! 

I'm looking at ROS integration, did you manage to get it all working?

Cheers.

bxs1260, 2016-05-12:
@LovelyHorse Thanks a lot Keaton for explaining this important fix. This fix simply works :+1:

ksze, 2016-10-03:
Please bear with me if I sound like a complete noob. But what is this NiTE thing that you are talking about? If I just run NiViewer2 like the [README.md] says, I don't get any skeleton tracking. What should I be expecting to see in NiViewer2?

rafapages, 2016-10-03:
NiTE is the library you need to find the skeleton with the data Kinect provides. NiViewer2 gives you access to the different Kinect streams: depth and color, basically.

josuerocha, 2016-12-28:
Can somebody please help me to install OpenNi 2 with Kinect Sensors in Ubuntu? I am trying to install but I get this error when installing SensorKinect :

../../../../Source/XnCore/XnCoreStatus.cpp:26:30: fatal error: XnStatusRegister.h: No such file or directory
 #include <XnStatusRegister.h>

please help me, thanks in advance.

fma-1724, 2017-07-19:
Hi!

I'm doing the final project of my degree and i need to know if you can use the kinect v2 with OpenNI and NiTE, using Ubuntu (I can't say the specific version because my current computer don't have 3.0 USB and they haven't lend me a computer with it yet).

For what i understand of your conversation there is a way so if any of you could explain or reference me to where i can got the answer i would be really grateful.

Thanks