OpenKinect/libfreenect2 #633: Failure when `make install` [Ubuntu 15.04]

zry656565, 2016-04-12:
- Platform: Ubuntu 15.04
- Video card: Nvidia GeForce 860M

I installed dependencies accordding to the guide for ubuntu 14.04, and libusb & OpenGL were installed well.
Then I make libfreenect2 successfully, but met some problem when `make install`.

**Result of `cmake`**

```bash
jerry@tu1504:~/libs/libfreenect2/build$ cmake ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'libusb-1.0>=1.0.20'
--   found libusb-1.0, version 1.0.20
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0  
-- checking for module 'glfw3'
--   found glfw3, version 3.1.2
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "1.2") 
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
-- Found CUDA: /usr/local/cuda (found version "7.5") 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 pthread
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
[... 108 lines cut ...]
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jerry/libs/libfreenect2/build
[  2%] Built target generate_resources_tool
Linking CXX shared library lib/libfreenect2.so
[ 68%] Built target freenect2
Linking CXX shared library lib/libfreenect2-openni2.so
[ 91%] Built target freenect2-openni2
Linking CXX executable ../bin/Protonect
[100%] Built target Protonect
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /usr/local/lib/libfreenect2.so.0.2.0
CMake Error at cmake_install.cmake:48 (file):
  file INSTALL cannot copy file
  "/home/jerry/libs/libfreenect2/build/lib/libfreenect2.so.0.2.0" to
  "/usr/local/lib/libfreenect2.so.0.2.0".

Makefile:66: recipe for target 'install' failed
make: *** [install] Error 1
```

Any idea for this?

ThomasMoinel, 2016-04-12:
/usr/local/lib/ can only be written by root.

You should execute you command with sudo:
`sudo make install`

zry656565, 2016-04-12:
@ThomasMoinel 
Excellent explaination! The error has been fixed, thanks!

closed 2016-04-12 by zry656565