OpenKinect/libfreenect2 #634: unable to run kinect v2 using libfreenect2

nomi321, 2016-04-12:
I am unable to run kinect v2 using libfreenect2. after following all the steps mentioned and after running this command i got the following errors. Kindly help me. What can be the reason

noaman@Rime:~/libfreenect2/build$ ./bin/Protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:19 with serial 160634134647
[Info] [Freenect2Impl] found 1 devices
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Mobile - 1.7.0
[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:19
failure opening device!

shamoorti, 2016-04-12:
Have you tried sudo ./bin/Protonect?

nomi321, 2016-04-12:
no i try this but i think this error has nothing to do with sudo. don't you think so ?

[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Mobile - 1.7.0[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBUSB_DEBUG=3

shamoorti, 2016-04-12:
Currently I'm having problem with Vappi access. Because when I use sudo to run protonect Vaapi works otherwise it wont. What does you -vainfo says? Also it seems you have pluged the kinect to USB 2.0. Your USB error is because of that.

nomi321, 2016-04-12:
and even trying that it gave same output ...

    On Tuesday, April 12, 2016 7:48 PM, Noaman Mehmood <enggr_nomi@yahoo.com> wrote:
 

 no i try this but i think this error has nothing to do with sudo. don't you think so ?

[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Mobile - 1.7.0[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBUSB_DEBUG=3

shamoorti, 2016-04-12:
First of all print your out put here of vainfo. Second of all you have plugged your kinect to the wrong USB port. It should be a USB 3.0 port. Did you change your port?

nomi321, 2016-04-12:
when I run sudo ./bin/Protonect
i get the following output 
noaman@Rime:~/libfreenect2/build$ sudo ./bin/Protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:27 with serial 160634134647
[Info] [Freenect2Impl] found 1 devices
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
[Warning] [ShaderProgram] Working around buggy boolean instructions in your Mesa driver. Mesa DRI 10.3+ is recommended.
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Mobile - 1.7.0
[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:27
failure opening device!
now I m not sure about r u asking about print output of vainfo ? 
and secondly ok m trying to replace usb port but what if I dont have usb 3.0 port ?

nomi321, 2016-04-12:
and i just checked that my system dont have usb 3.0 port. i was not aware of the issue too ...

nomi321, 2016-04-12:
but if it so then how detected the sensor as you can see from its output that 
[Info] [Freenect2Impl] found valid Kinect v2 @2:27 with serial 160634134647

what does it mean ? now what i have to do please let me know

shamoorti, 2016-04-12:
It just means that USB can recognize the Kinect, but it doesn't mean it can communicate with it and transfer the data. You absolutely need USB 3.0 port. That's the prerequisite of Kinect V2.

nomi321, 2016-04-12:
ok i solve it and what about the other error as output was showing two errors ?
[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported

shamoorti, 2016-04-12:
Print your Protonect log scene while you run it to see if Vaapi is working now. Run ./bin/Protonect.

nomi321, 2016-04-12:
i didnt get your point. I have already shown you the whole output when i run the above mentioned command. can u elaborate your point please ?

nomi321, 2016-04-12:
plus I am studying that there can be issue of graphics card. I need to update my system now for this. ufff. is there specific requirments of graphics card too to run the sensor ?

shamoorti, 2016-04-12:
You can use both Nvidia Cuda and Intel OpenCL accelerators. It depends on what you want to do. If you have heavy processing and high level image processing definitely you need better GPUs even double of them. First evaluate what is the purpose of your application and Kinect.

xlz, 2016-04-12:
It seems you didn't install udev rules.

VAAPI is not supported in SandyBridge - read the README https://github.com/OpenKinect/libfreenect2#requirements-for-optional-features. This problem is irrelevant for you.

nomi321, 2016-04-13:
i want to use kinect in order to identify objects .. 3D object identification is my main aim .... I think in my system it is sandybridge (intel HD graphics Family). Is it good enough ?

nomi321, 2016-04-13:
as you can see it is shown above 

[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Sandybridge Mobile - 1.7.0

xlz, 2016-04-16:
VAAPI is not supported on SandyBridge.

closed 2016-04-16 by xlz

whatisor, 2017-02-09:
Just for reference,
I got same issue 
"Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBUSB_DEBUG=3 ."

WHEN Kinect cable and Kinect for Window HUB connection is not firm.

jiapei100, 2019-07-29:
@nomi321 
I got exactly the same problem... How did you solve it??

Thank you 
Pei