OpenKinect/libfreenect2 #636: Running libfreenect 2 on Intel NUC

pnambiar, 2016-04-12:
I'm trying to run libfreenect 2 on Inel NUC -NUC5i7RYH with ubuntu Ubuntu 14.04.4 LTS.
I was able to install it successfully. However ./Protonect gives me the following error
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @3:3 with serial 503931742542
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 X11: RandR gamma ramp support seems broken
Xlib:  extension "NV-GLX" missing on display ":0".
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.

Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

xlz, 2016-04-12:
`./bin/Protonect -noviewer cl`

pnambiar, 2016-04-13:
./bin/Protonect -noviewer cl
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
X server found. dri2 connection failed! 
X server found. dri2 connection failed! 
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Error] [OpenCLDepthPacketProcessorImpl] could not find any suitable device
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
Segmentation fault (core dumped)

pnambiar, 2016-04-13:
After installing beignet1.1. I get

./bin/Protonect -noviewer cl
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
X server found. dri2 connection failed! 
Device open failed, aborting...
X server found. dri2 connection failed! 
Device open failed, aborting...
cl_get_gt_device(): error, unknown device: ffffffff
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Error] [OpenCLDepthPacketProcessorImpl] could not find any suitable device
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34
[Error] [OpenCLAllocator] b->buffer = cl::Buffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR, size, NULL, &err): -34

wiedemeyer, 2016-04-14:
looks like the same problem as here: https://github.com/code-iai/iai_kinect2/issues/248
But that was with a Skylake NUC, not the Broadwell NUC.

xlz, 2016-04-22:
There is no much we can do here because this is really a system configuration issue.

closed 2016-12-05 by xlz