OpenKinect/libfreenect2 #637: Loss of Depth Packets

niteshtm, 2016-04-12:
Hi there!! I have built libfreenect2 with OpenCL . But in the console log I can see loss of Depth packets like 30 packets lost sometimes 15 packets lost . I have Intel i5 with onboard Intel graphics card ,using  usb 3.0 control . My total CPU usage is not exceeding 50-60 % . And I can notice frame drops . Just wanted to know how these packet loss can be decreased . How can i get 30 fps ?

xlz, 2016-04-12:
`./bin/Protonect -noviewer cl`.

niteshtm, 2016-04-13:
Sorry ,l I didn't get you .

hanshammel1337, 2016-04-14:
He asked you to post the output of `Protonect` with the arguments `-noviewer cl`. User xlz assumes your are using Linux, so he posted the Linux command for this. As your screenshots show, your are using Windows. I'am not familiar with libfreenect2 and Windows, but in your 1st screenshot you see the avaible arguments for KinectStreams.exe (which seems the same as Protonect in the Linux world?!). So just run `KinectStreams.exe -noviewer cl` and post the output here

niteshtm, 2016-04-14:
Thanks hanshammel1337 !! Well even exclusively providing that argument also I am getting the same result only. I am using Windows 8.1 to be very specific.

xlz, 2016-04-16:
That's the best you can get because the lower performance of JPEG decoding on Windows. If you want better performance use Linux.

closed 2016-04-16 by xlz

niteshtm, 2016-04-16:
That's sad !! My application is on Windows only :(