OpenKinect/libfreenect2 #638: How to create two seperate listeners for rgb and depth data?

niteshtm, 2016-04-13:
Need to create two seperate listeners for rgb and depth data? How can we implement it? 
Thanks.

xlz, 2016-04-16:
```
libfreenect2::SyncMultiFrameListener rgb_listener(libfreenect2::Frame::Color);
libfreenect2::SyncMultiFrameListener ir_listener(libfreenect2::Frame::Ir | libfreenect2::Frame::Depth);
dev->setColorFrameListener(&rgb_listener);
dev->setIrAndDepthFrameListener(&ir_listener);
```

closed 2016-04-16 by xlz

niteshtm, 2016-04-16:
No but I thought SyncMultiFrameListener still waits for all the frames to arrive as it works in sync and that causes a very small amount of  delay . I was not aware of this fact that if we pass Color only as argument then it won't wait for others frame internally or so ? 
Correct me if I am wrong please .

niteshtm, 2016-04-28:
Still waiting for an answer.

xlz, 2016-04-28:
You were not wrong, so I didn't correct you.

niteshtm, 2016-04-28:
xlz: I have calculated the fps I am getting and it's really bad. On an average it's 8 . Please suggest me something in order to increase the fps if there is any way.

xlz, 2016-04-28:
Use one of the better configurations:
[image]

niteshtm, 2016-04-28:
Means I can't get better fps,since I have i5 , and using Opencl .

xlz, 2016-04-28:
I don't know your hardware. You may be doing something wrong to be slower than ARM CPU.

niteshtm, 2016-04-28:
Processor : Intel core i5-3470 CPU@3.20 Ghz
Ram- 4GB
OS : 8.1(64bit)
GPU : On board Intel Graphics card.
 
[image]

xlz, 2016-04-28:
8 fps is the expected performance for the step. If you want better use Linux.

niteshtm, 2016-04-28:
Thanks man for the confirmation . :)