OpenKinect/libfreenect2 #639: Fail to use libfreenect2 with OpenNI2 [Ubuntu 15.04]

zry656565, 2016-04-14:
- Platform: Ubuntu 15.04
- Video card: Nvidia GeForce 860M

I wanna use libfreenect2 with OpenNI2 built by myself.

- Location of **OpenNI2**: `/home/jerry/lab/Kintinuous/deps/OpenNI2/Bin/x64-Release/`
- Location of **libfreenect2**: `/home/jerry/libs/libfreenect2`
- Location of **freenect2** after `make install`: `/home/jerry/libs/freenect2`

**`libfreenect2/build/bin/Protonect` works well.**

```
jerry@tu1504:~/libs/libfreenect2/build/bin$ ./Protonect 
X server found. dri2 connection failed! 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:7 with serial 021674143547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 021674143547
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing
```

I tried to create a soft link like blew:
```
jerry@tu1504:~/libs/freenect2/lib/OpenNI2/Drivers$ ln -s libfreenect2-openni2.so ~/lab/Kintinuous/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers/
jerry@tu1504:~/libs/freenect2/lib/OpenNI2/Drivers$ ln -s libfreenect2-openni2.so.0 ~/lab/Kintinuous/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers/
```

And `NiViewer` still could not detect the Kinect v2 device.
```
jerry@tu1504:~/lab/Kintinuous/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers$ ../../NiViewer 
openDevice failed:
	DeviceOpen using default: no devices found

Press any key to continue . . .
```

I didn't find some useful information for me in #243, #311, #523. Did I miss something to solve this problem?

xlz, 2016-04-16:
I don't think they way you create symlinks is correct. Check that again.

Also I'm not sure of the path you used here `OpenNI2/Bin/x64-Release/OpenNI2/Drivers/`. You need to make sure if this is the right path for installing drivers. The path should be specified in OpenNI.ini.

closed 2016-04-16 by xlz

hanyazou, 2016-04-16:
@zry656565 did you run `sudo make install-openni2` ?

zry656565, 2016-04-18:
@xlz 

You are right, so I recreate symlinks with absolute path to avoiding Error **'Too many levels of symbolic links'**.

```bash
jerry@tu1504:~/libs/freenect2/lib$ ln -s /home/jerry/libs/freenect2/lib/OpenNI2/Drivers/libfreenect2-openni2.so /home/jerry/lab/Kintinuous/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers/
jerry@tu1504:~/libs/freenect2/lib$ ln -s /home/jerry/libs/freenect2/lib/OpenNI2/Drivers/libfreenect2-openni2.so.0 /home/jerry/lab/Kintinuous/deps/OpenNI2/Bin/x64-Release/OpenNI2/Drivers/
```

I didn't modify `OpenNI.ini`. There two `OpenNI.ini` in my system:
- `/etc/openni2/OpenNI.ini`
- `/home/jerry/lab/Kintinuous/deps/OpenNI2/Config/OpenNI.ini`

The content of these two files are same like below. I don't know which takes effect.

```ini
[Log]
; 0 - Verbose; 1 - Info; 2 - Warning; 3 - Error. Default - None
Verbosity=3
LogToConsole=0
LogToFile=0

[Device]
;Override=""

[Drivers]
; Location of the drivers specified by a relative path based on OpenNI's shared library or an absolute path.
; Path separator "/" can be used to be portable for any platforms.
; Default - OpenNI2/Drivers
;Repository=OpenNI2/Drivers
```

So I tried to create symlinks of libfreenect2-openni2.so to `/etc/openni2/OpenNI2/Drivers`, but that didn't work.

@hanyazou 

Yes, so I get the folder `OpenNI2/Driver` in `/home/jerry/libs/freenect2`.

```bash
jerry@tu1504:~/libs/libfreenect2/build$ sudo make install-openni2
[sudo] password for jerry: 
[  3%] Built target generate_resources_tool
[ 75%] Built target freenect2
[100%] Built target freenect2-openni2
[100%] Built target install-openni2
jerry@tu1504:~/libs/freenect2/lib/OpenNI2/Drivers$ cd ~/libs/freenect2/lib/OpenNI2/Drivers/
jerry@tu1504:~/libs/freenect2/lib/OpenNI2/Drivers$ ls
libfreenect2-openni2.so  libfreenect2-openni2.so.0
```

robnsngh, 2016-04-27:
@zry656565 
I just compiled freenect2 with OpenNI2 support (on Ubuntu 14.04).
my OpenNI2 is located in /usr and I had the similar problem. 
For sake of simplicity I built libfreenect2 into /usr and it worked fine with it. Its untidy but it worked. 
Looks like you might have OpenNI2 installed in /usr so try giving that a shot ;) 

I am also trying to compile Kintinuous and after OpenNI2 port + changing the stream resolution to 512,424 Kintinuous is running (not perfectly though)

xlz, 2016-04-27:
Seriously, I've never been able to reproduce these issues. The README instructions say to install OpenNI2 from apt-get and its location would be in /usr/lib. If you follow the README verbatim, I don't believe you'll be able to reproduce these issues.

But this is not really a hard problem. Change OpenNI.ini to
```
[Log]
; 0 - Verbose; 1 - Info; 2 - Warning; 3 - Error. Default - None
Verbosity=0
LogToConsole=1
LogToFile=0
```
run NiViewer2 and see the console output, which will show you the problem. If it is still not informative, `strace -etrace=open -f -o/tmp/trace NiViewer2`, and examine `/tmp/trace` to look for `open(".../libfreenect2-openni2.so"...) = -1 (ENOENT)`, and the problem solves itself!

zry656565, 2016-06-06:
@robnsngh @xlz 
Finally, I made it work following your advises, thanks!

mathemaphysics, 2021-05-17:
I don't know if this might be useful or not, but I had significant issues getting the libFreenectDriver.so to be found and just solved the problem by setting the permissions and link names correctly for `libFreenectDriver.so` and its necessary link `libFreenectDriver.so.0`. These names are non-negotiable with OpenNI 2 apparently. Maybe this was already known by everyone, but I did everything else the recommended way but this.

Taking the advice of @xlz and running `stract` but instead with `-etrace=openat -f -o /tmp/trace` showed that it wasn't even opening the shared library at all. It wasn't even getting a mention even though it was in `/usr/lib/OpenNI2/Drivers` as it should have been. I ran `chmod 644` on the shared library and created a symlink to the right name suffixed with a `.0` and it worked.