OpenKinect/libfreenect2 #640: Ubuntu 14.04 - Protonect not showing depth images with cl or gl

hanshammel1337, 2016-04-14:
hello,

the viewer from protonect is not showing any depth images when i use arguments `cl` oder `gl`.
- `./Protonect cpu` --> viewer shows rgb camera image and depth images
- `./Protonect gl` --> viewer shows only rgb camera image (bottom left corner)
- `./Protonect cl` --> viewer shows only rgb camera image (bottom left corner)

i'am using a notebook with hybrid graphics. but for now ubuntu is forced to use the dedicated amd/ati graphic card with proprietary `fglrx` driver.
when i run `Protonect` with argument `-noviewer cl` it seems to work:
```
[Info] [OpenCLDepthPacketProcessor] avg. time: 5.9364ms -> ~168.452Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 23.3443ms -> ~42.837Hz
The viewer is turned off. Received 1200 frames. Ctrl-C to stop.
[Info] [OpenCLDepthPacketProcessor] avg. time: 5.86356ms -> ~170.545Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 22.9544ms -> ~43.5646Hz
The viewer is turned off. Received 1300 frames. Ctrl-C to stop.
```
and same with argument `-noviewer gl`:
```
[Info] [OpenGLDepthPacketProcessor] avg. time: 10.882ms -> ~91.8953Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 24.0571ms -> ~41.5677Hz
The viewer is turned off. Received 600 frames. Ctrl-C to stop.
[Info] [OpenGLDepthPacketProcessor] avg. time: 10.5617ms -> ~94.6821Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 23.6082ms -> ~42.3582Hz
The viewer is turned off. Received 700 frames. Ctrl-C to stop.
```
any suggestions or other help why the viewer is not displaying the depth images, when the computation seems ok?

xlz, 2016-04-16:
Computation is OK as long as you have framerate information, like these
```
[Info] [OpenCLDepthPacketProcessor] avg. time: 5.86356ms -> ~170.545Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 10.5617ms -> ~94.6821Hz
```

The viewer doesn't work probably because something is wrong with your graphics driver.

closed 2016-04-16 by xlz

hanshammel1337, 2016-04-18:
The graphic card driver seems to be installed correctly. Applications like `glxgears`, `glmark2 and `fgl_glxgears` are working as expected. Even a installed test game from official repository `foobillardplus` is running without any issues.

I reinstalled some packages and made a clean new build. Now I get everything displayed correctly with `./Protonect cl`, but `./Protonect gl` has so same behaviour like before.