OpenKinect/libfreenect2 #642: Error when I try to run NiViewe2 Ubuntu 14.0.4

aaramirezd, 2016-04-15:
Hi everybody,

I got an error when I try to run NiViewe2 but Protonect run well

Ubuntu 14.0.4:
Nvidia Geforce
/home/andres/libfreenect2
/home/andres/freenect2

andres@andres:~/libfreenect2/build$ NiViewer2 
openDevice failed:
	DeviceOpen using default: no devices found

Press any key to continue . . .

Thanks
Andrés

xlz, 2016-04-16:
Not enough information...

aaramirezd, 2016-04-18:
Ubuntu 14.04.4 LTS:
Kernel: 4.2.0.35 X86_64
GeForce GT 730 @ 1400MHz Memory 2047MB 
/home/andres/libfreenect2
/home/andres/freenect2
libcudart.so.7.5 in the LD_LIBRARY_PATH

I have a problem with libva

**./bin/Protonect gl**

andres@andres:~/libfreenect2/build$ ./bin/Protonect gl
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 038680345147
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 038680345147
device firmware: 4.0.3914.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 992
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 9.69083ms -> ~103.19Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.7983ms -> ~67.5753Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.44257ms -> ~118.447Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.9242ms -> ~67.0055Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.69887ms -> ~114.957Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.7545ms -> ~67.776Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.70719ms -> ~114.848Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.573ms -> ~68.6199Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.89427ms -> ~112.432Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.5915ms -> ~68.5331Hz

**./bin/Protonect cl**

andres@andres:~/libfreenect2/build$ ./bin/Protonect cl
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
[... 71 lines cut ...]
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 30 packets were lost
[Debug] [DepthPacketStreamParser] skipping depth packet

chaohuig, 2016-04-21:
First of all, @xlz  thanks for your wonderful work and sharing. 
I am experiencing exactly the same problem as @aaramirezd did. Do you have any suggestion on how to proceed? Thanks in advance.

xlz, 2016-04-27:
`LD_LIBRARY_PATH=/usr/local/cuda/lib64 NiViewer2`. This LD_LIBRARY_PATH is supposed to be set by the end-user. I can't put it in the code.

VAAPI does not work if it is not supported by your CPU or you didn't install respective drivers, I don't know.

closed 2016-04-27 by xlz

twmht, 2016-06-24:
hi @xlz ,

Where is the source code of NiViewr2?

I did not find that from the source code of libfreenect2.