OpenKinect/libfreenect2 #644: Skeleton Data from OpenNI2

kerolos, 2016-04-18:
hello Everyone,

I had installed openni2, and it works fine. I do nott know how can I get the skeleton data, that describes the location  and identifies the positions of the joints. or I should install other libraries such as PrimeSense or NITE. I am using ubuntu operating system. 
Anther question, it is better if I use SDK in windows that Microsoft Kinect provided to get better results.

I really appreciate any answer, Thank You in advance

xlz, 2016-04-19:
If you want skeleton data, I think the Microsoft SDK is better for you because it is not straightforward to get skeleton working with libfreenect2.

closed 2016-04-19 by xlz

kerolos, 2016-04-19:
actually,Now I am using ubuntu os and I wanna work with PCL in the future. 
Could you give me a hint, how can i get Skeleton data by using these libraries "libfreenet2, openni2 .."
thanks in advance

HenningJ, 2016-04-19:
Yes, using the libfreenect2 OpenNi driver combined with NiTE, you could get skeleton data. That does work. But you are not really allowed to use NiTE with non-PrimeSense devices.

rafapages, 2016-04-19:
Check out #243 

There is info about how to setup OpenNI and NiTE.