OpenKinect/libfreenect2 #649: GLFW error 65543 + Failed to create opengl window

jacksonkr, 2016-04-27:
`rtabmap` works fine with kinect v1 but not kinect v2. Both `Protonect cl` and `Protonect cpu` work but `Protonect gl` fails to render depth so you only see IR and RGB. I found a [similar issue](https://github.com/OpenKinect/libfreenect2/issues/409) but the OP was running Nvidia and was having a driver issue.

```
$ rtabmap
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
jackson@jackson-MacBookPro:~/rtabmap/build$ glxinfo | grep "OpenGL version"
OpenGL version string: 4.5.13399 Compatibility Profile Context 15.201.1151

$ lspci | grep "VGA"
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Venus XT [Radeon HD 8870M / R9 M270X/M370X] (rev 83)

$ fglrxinfo
display: :0  screen: 0
OpenGL vendor string: Advanced Micro Devices, Inc.
OpenGL renderer string: AMD Radeon R9 (TM) M370X
OpenGL version string: 4.5.13399 Compatibility Profile Context 15.201.1151

$ dmesg | grep fglrx | grep module
[    3.935073] fglrx: module license 'Proprietary. (C) 2002 - ATI Technologies, Starnberg, GERMANY' taints kernel.
[    3.939210] fglrx: module verification failed: signature and/or  required key missing - tainting kernel
[    3.973826] <6>[fglrx] module loaded - fglrx 15.20.3 [Sep  8 2015] with 1 minors
```

mbp2015, Ubuntu 14.04 x64

xlz, 2016-04-27:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#opengl-only-check-driver-version

jacksonkr, 2016-04-27:
@xlz OpenGL is > 3.1 in both cases:

> (OpenGL only) Check driver version
> 
> glxinfo | grep OpenGL
> 
> Either the "OpenGL core profile version string" or the "OpenGL version string" must be greater than 3.1. If not, you don't have required graphics driver or you have not installed one correctly. Also, you should not see llvmpipe which is a software OpenGL renderer.
> 
> Insufficient version may appear as GLFW error 65543 The requested client API version is unavailable.

```
$ glxinfo | grep OpenGL
OpenGL vendor string: Advanced Micro Devices, Inc.
OpenGL renderer string: AMD Radeon R9 (TM) M370X
OpenGL core profile version string: 4.3.13399 Core Profile Context 15.201.1151
OpenGL core profile shading language version string: 4.40
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.5.13399 Compatibility Profile Context 15.201.1151
OpenGL shading language version string: 4.40
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
$ glxinfo | grep llvmpipe
$ 
```

xlz, 2016-04-28:
OK this is new.

Try this:

1. `cd depends; ./install_glfw.sh`
2. `./glfw_src/build/tests/glfwinfo -m 3 -n 1`

Post the output.

jacksonkr, 2016-04-28:
```
$ cd depends; ./install_glfw.sh
Cloning into '/home/jackson/Downloads/libfreenect2/depends/glfw_src'...
remote: Counting objects: 17897, done.
remote: Total 17897 (delta 0), reused 0 (delta 0), pack-reused 17897
Receiving objects: 100% (17897/17897), 8.84 MiB | 1.99 MiB/s, done.
Resolving deltas: 100% (11979/11979), done.
Checking connectivity... done.
Note: checking out '3.0.4'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at eab31f2... Added description of 3.0.4 release.
-- The C compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
-- Using X11 for window creation
-- Using GLX for context creation
-- Looking for glXGetProcAddress
-- Looking for glXGetProcAddress - found
-- Looking for glXGetProcAddressARB
-- Looking for glXGetProcAddressARB - found
-- Looking for glXGetProcAddressEXT
-- Looking for glXGetProcAddressEXT - not found
-- Configuring done
CMake Warning (dev) in src/CMakeLists.txt:
  Policy CMP0022 is not set: INTERFACE_LINK_LIBRARIES defines the link
  interface.  Run "cmake --help-policy CMP0022" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  Target "glfw" has an INTERFACE_LINK_LIBRARIES property which differs from
  its LINK_INTERFACE_LIBRARIES properties.

  INTERFACE_LINK_LIBRARIES:

    /usr/lib/x86_64-linux-gnu/libX11.so;-lpthread;/usr/lib/x86_64-linux-gnu/libXrandr.so;/usr/lib/x86_64-linux-gnu/libXi.so;/usr/lib/x86_64-linux-gnu/libXxf86vm.so;/usr/lib/x86_64-linux-gnu/librt.so;/usr/lib/x86_64-linux-gnu/libm.so;/usr/lib/x86_64-linux-gnu/libGL.so

[... 160 lines cut ...]
-- Install configuration: ""
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/include/GLFW
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/include/GLFW/glfw3native.h
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/include/GLFW/glfw3.h
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/lib/cmake/glfw/glfwConfig.cmake
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/lib/cmake/glfw/glfwConfigVersion.cmake
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/lib/cmake/glfw/glfwTargets.cmake
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/lib/cmake/glfw/glfwTargets-noconfig.cmake
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/lib/pkgconfig/glfw3.pc
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/lib/libglfw.so.3.0
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/lib/libglfw.so.3
-- Installing: /home/jackson/Downloads/libfreenect2/depends/glfw/lib/libglfw.so
jackson@jackson-MacBookPro:~/Downloads/libfreenect2/depends$ ./glfw_src/build/tests/glfwinfo -m 3 -n 1
GLFW header version: 3.0.4
GLFW library version: 3.0.4
GLFW library version string: "3.0.4 X11 GLX glXGetProcAddress clock_gettime /dev/js shared"
OpenGL context version string: "3.1.13399 Compatibility Profile Context 15.201.1151"
OpenGL context version parsed by GLFW: 3.1.13399
OpenGL context flags (0x00000000):
OpenGL context flags parsed by GLFW: debug
OpenGL context renderer string: "AMD Radeon R9 (TM) M370X"
OpenGL context vendor string: "ATI Technologies Inc."
OpenGL context shading language version: "4.40"

```

xlz, 2016-04-28:
GLFW has no problem with GL 3.1, then what is the problem??? The above did test the same thing libfreenect2 does here https://github.com/OpenKinect/libfreenect2/blob/master/src/opengl_depth_packet_processor.cpp#L869-L887

Give me
```
dpkg -l libglfw*
ldd bin/Protonect | grep glfw
```

Also, in the build directory, remove everything and rebuild and try again?

jacksonkr, 2016-04-28:
```
$ dpkg -l libglfw*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
un  libglfw-dev    <none>       <none>       (no description available)
ii  libglfw3:amd64 3.1.2-3      amd64        portable library for OpenGL, wind
ii  libglfw3-dev:a 3.1.2-3      amd64        portable library for OpenGL, wind
$ ldd bin/Protonect | grep glfw
	libglfw.so.3 => /usr/lib/x86_64-linux-gnu/libglfw.so.3 (0x00007feb47b90000)

```

xlz, 2016-04-28:
OK

1. Edit `install_glfw.sh` and change "git checkout 3.0.4" to "git checkout master", which should report more descriptive errors for version problem.
2. `glfw_src/build/tests/glfwinfo -m 3 -n 1`?
3. ... `./bin/Protonect`?

jacksonkr, 2016-04-28:
```
$ ./install_glfw.sh 
Cloning into '/home/jackson/Downloads/libfreenect2/depends/glfw_src'...
remote: Counting objects: 17897, done.
remote: Total 17897 (delta 0), reused 0 (delta 0), pack-reused 17897
Receiving objects: 100% (17897/17897), 8.84 MiB | 1.81 MiB/s, done.
Resolving deltas: 100% (11979/11979), done.
Checking connectivity... done.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
-- The C compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find Vulkan (missing:  VULKAN_LIBRARY VULKAN_INCLUDE_DIR) 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
-- Using X11 for window creation
-- Using GLX for context creation
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jackson/Downloads/libfreenect2/depends/glfw_src/build
Scanning dependencies of target glfw
[  1%] Building C object src/CMakeFiles/glfw.dir/context.c.o
[  2%] Building C object src/CMakeFiles/glfw.dir/init.c.o
[  3%] Building C object src/CMakeFiles/glfw.dir/input.c.o
[  5%] Building C object src/CMakeFiles/glfw.dir/monitor.c.o
[  6%] Building C object src/CMakeFiles/glfw.dir/vulkan.c.o
[  7%] Building C object src/CMakeFiles/glfw.dir/window.c.o
[  8%] Building C object src/CMakeFiles/glfw.dir/x11_init.c.o
[ 10%] Building C object src/CMakeFiles/glfw.dir/x11_monitor.c.o
[ 11%] Building C object src/CMakeFiles/glfw.dir/x11_window.c.o
[ 12%] Building C object src/CMakeFiles/glfw.dir/xkb_unicode.c.o
[ 14%] Building C object src/CMakeFiles/glfw.dir/linux_joystick.c.o
[ 15%] Building C object src/CMakeFiles/glfw.dir/posix_time.c.o
[ 16%] Building C object src/CMakeFiles/glfw.dir/posix_tls.c.o
[ 17%] Building C object src/CMakeFiles/glfw.dir/glx_context.c.o
Linking C shared library libglfw.so
[ 17%] Built target glfw
Scanning dependencies of target boing
[ 19%] Building C object examples/CMakeFiles/boing.dir/boing.c.o
[ 20%] Building C object examples/CMakeFiles/boing.dir/__/deps/glad.c.o
Linking C executable boing
[ 20%] Built target boing
Scanning dependencies of target gears
[ 21%] Building C object examples/CMakeFiles/gears.dir/gears.c.o
[ 23%] Building C object examples/CMakeFiles/gears.dir/__/deps/glad.c.o
Linking C executable gears
[... 175 lines cut ...]
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Error] [VaapiRgbPacketProcessorImpl] vaDisplayIsValid(display) failed
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 038091735247
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 038091735247
device firmware: 4.0.3912.0
```

I'm not sure if you wanted me to re`make` after `install_glfw.sh` so I tried without re`make` and the tried again with re`make`.

xlz, 2016-04-28:
You don't need to should the output of install_glfw.sh which has no relevant information.

> so I tried without remake and the tried again with remake.

I don't know what this means.

So in the last output of `./bin/Protonect gl`, did it work or not?

xlz, 2016-04-28:
Running `install_glfw.sh` will build a local copy of glfw in the depends directory and if you then build libfreenect2 from scratch, this local copy in the directory will be preferred over the system glfw. If you don't want libfreenect2 to link to this local copy of glfw you need to remove glfw from the depends directory.

I was asking you to try different versions of glfw in this way.

With the new local copy of glfw, what errors do rtabmap show? The error should be in this format: `Requested client API version %i.%i, got version %i.%i`.

> Protonect gl fails to render depth so you only see IR and RGB

It's not very clear what this means. Did you mean `Protonect gl` show a window that visualizes the ir and rgb images, but not the depth image and the registered image?

jacksonkr, 2016-04-28:
Yes, I could have phrased that better. What I mean is that the right two visuals (top-right & bottom-right) are not visible with `Protonect gl`. I'm not exactly sure what the name of those visuals are but to compare, all visuals work with both `cpu` and `cl`.

```
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 The requested client API version is unavailable
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
```

Again, that's running `rtabmap` and directing it to use `Freenect2`. Once `Freenect2` is selected the `rtabmap` app crashes with the above error. `libfreenect` works for Kinect 360 and `OpenNi` but that's it for now.

xlz, 2016-04-29:
`apt-get remove libglfw3` and then do the thing again? This is to make sure rtabmap is linked to the local glfw master (`ldd rtabmap | grep glfw` shows `.../depends/glfw/lib/libglfw.so`), which is not the case from the error output. This is for getting the version information in `equested client API version %i.%i, got version %i.%i`.

xlz, 2016-04-29:
If you just care about having something that works, `cmake -DENABLE_OPENGL=OFF` to rtabmap will stop using the OpenGL pipeline. (But this bug remains.)

jacksonkr, 2016-04-29:
- _made_ `freenect2`
- _made_ `rtabmap`
- _ran_ `rtabmap`

```
$ rtabmap
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 Requested client API version 3.1, got version 2.1
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
```

At least I'm getting an API version now. `Requested client API Versions 3.1, got version 2.1`

xlz, 2016-04-29:
I know this is a bug, but I can't guess anything more.

jacksonkr, 2016-04-29:
@xiz Okay, thanks for your help. I'll post back when I figure out what works.

jacksonkr, 2016-05-02:
I tried a number of things to get this to work on my system but what I ended up doing was installing a fresh copy of ubuntu 14.04.3 x64, installed `ros-indigo` then built `libfreenectv2`. GL now works as expected.

closed 2016-05-02 by jacksonkr

xlz, 2016-05-02:
Mystical driver problem, always like this.