OpenKinect/libfreenect2 #650: Black dot in depth image

AlexxxBeck, 2016-04-28:
Hello guys,

I have a problem with the depth image of the kinect 2. This is just an example of a screenshot taken with the rqt_image_viewer of the depth image:
[image]

So the depth image always contains the dark "dot" but I don't know where it comes from. So also when I'm using Protonect this black dot is showing. The thing is that I have two different Kinect's and tested it on different computers and I always get that black dot in the depth image. On one of the Kinect's the dot is stronger than on the other  but I unfortunately have no idea how I can get rid of it because the rest of the depth image works fine.
I would be really glad if you could help me.

Thank you very much, 
Alex

xlz, 2016-04-28:
Where in the image is the dark dot you refer to?

AlexxxBeck, 2016-04-28:
Here:
[image]

xlz, 2016-04-28:
It may be the surface is reflective and the area you marked reflects directly and saturates the measurements.

AlexxxBeck, 2016-04-28:
Yes the surface is reflective and I just tried the depth image on that specific part. I'll try again tomorrow on non-reflective parts. Thank you very much for the fast response.

AlexxxBeck, 2016-04-29:
So I just did some further tests and got the following results:
In this first image there is just a table without any objects and there is the clear black area in the picture:
[image]
So i thought I just bent the camera above that black dot and surprisingly it then disappeared althought it doesn't make a lot of sense to me why that should be.
[image]

Then i put an object below the camera (the same kinda reflective object I showed last time) and the big black area is still there:
[image]
so I thought perhaps it's really because the area of the surface might be reflective but when I move the object out of the black area the black are remains on exactly the same point like before:
[image]

Just for your notice these images were recorded with my other kinect so I doubt that the kinect is broken or something like that.

AlexxxBeck, 2016-04-29:
The problem with that is that in my program I create a difference image between the background and the foreground. So the difference image of my color stream is totally black after I use my program to get the difference image. 
Here is the depth image before my program:
[image]
and here it is after my program:
[image]

and as you can see in the are of the black are my difference image looks really weird and that's why I can't use it and try to solve it.

AlexxxBeck, 2016-04-29:
For your information in the pictures taken the camera was just around 50 cm away from the object. By increasing the distance between the camera and the objects the problem got slightly better but the black area is still there as you can see in this picture:
[image]

xlz, 2016-04-29:
What about the corresponding IR images?

AlexxxBeck, 2016-05-02:
So I could solve this issue by changing the camera to a vertical position and increasing the distance even further. Now everything works fine although I still don't know what caused this black area. Thank you really much for the fast responses though.

closed 2016-05-02 by AlexxxBeck