OpenKinect/libfreenect2 #651: openGL not installed

nomi321, 2016-04-30:
when i run this to install openGL 
sudo apt-get install libgl1-mesa-dri-lts-vivid
 i get the error

The following packages have unmet dependencies:
 unity-control-center : Depends: libcheese-gtk23 (>= 3.4.0) but it is not going to be installed
                        Depends: libcheese7 (>= 3.0.1) but it is not going to be installed

what is the reason ? how to solve this

xlz, 2016-04-30:
> (Ubuntu 14.04 only) sudo dpkg -i debs/libglfw3*deb; sudo apt-get install -f; sudo apt-get install libgl1-mesa-dri-lts-vivid (If the last command conflicts with other packages, don't do it.)

nomi321, 2016-05-01:
@xlz bro can you tell me that I want to use Kinect v2 for object identification ? which steps should I follow. Kindly help me if you can. I have installed its drivers and can run roslaunch successfully. Plz reply

xlz, 2016-05-02:
You can use opencv for object identification.

closed 2016-05-02 by xlz

nomi321, 2016-05-10:
open cv or pcl which one should I prefer for object identification and why ? please replay @xlz

xlz, 2016-05-10:
Mostly OpenCV does the job. But PCL can provide extra information with segmentation.

nomi321, 2016-05-10:
so it mean that i should go with PCL  ? @xlz