OpenKinect/libfreenect2 #652: OpenCL for Intel Sandybridge Processor

royalheadwear, 2016-05-02:
Hi,

I try to install the iai_kinect2. Therefore I need libfreenect2. I'm Using an XPS15 (L502X) with an Intel Core i7 2630QM and additional NVIDIA Geforce GT 540M graphics. Because my internal screen just works with the nouveau-driver (that's a different story), I can't install CUDA.

But when I Install OpenCL, like discribed, later the `clinfo` command throws me

`beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
E: -1`

I searched for a solution, but unfortunately I just found out, that the 2nd generation of Intel Core Processors is not supported by Beignet.

Can someone tell me a alternative for Beignet, that works for me? Is there a different way to get iai_kinect2 running? I think it uses OpenCL by default.

Thanks for your Help! And sorry If it's something like a stupid question, but I'm totaly new in ROS and using the Kinect Sensor and so on.

Have a nice evening ;)

xlz, 2016-05-02:
SandyBridge not having OpenCL support is documented in the README, if you've read it.

The best solution for you is to use Intel GPU for desktop and Nvidia for compute. (`apt-get install cuda; sudo prime-select intel`)

royalheadwear, 2016-05-03:
Thanks for the answer. 

At the Moment I'm Using my Laptop with an additional display on the HDMI-Port. The problem with that solution was, that the internal display of my Laptop gets black, when I select the NVidia GPU (but the HDMI-display works. And if I select the Intel GPU the HDMI-display goes to sleepmode and the internal display wakes up.

Is there a way, to use both displays independent from the GPU I'm using?

xlz, 2016-05-29:
You can switch the GPU back when you are not using Kinect.

closed 2016-05-29 by xlz