OpenKinect/libfreenect2 #656: Different threads for Depth and Color -> packet loss?

jmirats, 2016-05-13:
Hi,
I'm using two different threads to retrieve frames from Kinect2 sensor: One for Color frames, other for Ir and Depth frames, which are started from a main() program (very similar to what is done in iaikinect bridge from Wiedemeyer)

I'm a little bit puzzled since I'm experiencing random loss of frames either for depth or Color, and  actually need all of them. So I created buffers (std::vector) of cv::Mat objects to store all the frames that arrive, so in the main thread I can treat them independently of the time of arrival, but even in this way I'm loosing frames.

I have two listeners defined: listenColor and listenIrDepth

when measuring the time it takes for either 
listenColor->waitForNewFrame(frames);
or
listenIrDepth->waitForNewFrame(frames);

to return, I noticed sometimes is really big, anything above 70ms (and up to 500ms in the worst case I measured), which make me understand why I loose frames (usual delay for color and depth waits is 25 to 30ms, except when there is poor light which makes color wait drop down to 55-60 ms maintaining depth wait). But I do not understand what this is happening and how can I avoid it. It may be an effect of the two threads trying to access the USB? It is recommended to only have one thread for all frames?

Additional info:

I'm using Ubuntu 14.04, kernel 3.19.0-59-generic on a i7-4790 CPU at 3.6GHz with 8GB RAM.

Protonect works well with viewer:
[Info] [OpenGLDepthPacketProcessor] avg. time: 15.2502ms -> ~65.5729Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.4091ms -> ~64.8968Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.9162ms -> ~67.0413Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.4374ms -> ~64.7776Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 15.0905ms -> ~66.267Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.002ms -> ~66.6578Hz

or even better with -noviewer option:
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.8482ms -> ~67.3483Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 13.7237ms -> ~72.8669Hz
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.1718ms -> ~70.5629Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.6529ms -> ~68.246Hz
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
[Info] [OpenGLDepthPacketProcessor] avg. time: 14.3334ms -> ~69.7671Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 14.0837ms -> ~71.0041Hz

xlz, 2016-05-16:
Your cpu should support the OpenCL vaapi combination which should be much faster. Try it.

jmirats, 2016-05-16:
Thanks for the idea... I'll certainly try it. 
Will come back here when done, may be in a couple-three days

jmirats, 2016-05-20:
Hello, I finally checked with OpenCL and Vaapi. I still loose frames: at the beginning there are 13-14 frames lost, and after, I randomly loose frames (both, color and depth). This test is performed using a unique thread for receiving all the frames. 
In fact, I would say that OpenCL is slower than OpenGL in my computer.

Here is the output when running with OpenCL and Vaapi:

X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: GeForce GT 710 (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GT 710 (GPU)[NVIDIA Corporation]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 020346651447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Info] [DepthPacketStreamParser] -11 packets were lost

(Color Image:12547): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(Color Image:12547): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
(If you have multiple ICDs installed and OpenCL works, you can ignore this message)

(Color Image:12547): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(Color Image:12547): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(Color Image:12547): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [OpenCLDepthPacketProcessor] avg. time: 16.9721ms -> ~58.9201Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.113ms -> ~55.2089Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 16.9444ms -> ~59.0166Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.0904ms -> ~52.3824Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 16.9394ms -> ~59.0339Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.6719ms -> ~53.5564Hz
[Info] [OpenCLDepthPacketProcessor] avg. time: 17.0774ms -> ~58.5568Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.9924ms -> ~52.6528Hz

wiedemeyer, 2016-05-26:
Do you have the Intel HD GPU of your CPU disabled? VAAPI needs it. And with your Nvidia GPU you could also use the CUDA depth packet processor.

jmirats, 2016-05-27:
I do not know how to check if the Intel GPU is enabled or not.... however I would say that my computer is by default using the Nvidia. I have not checked with CUDA.

What I do not understand is the fact itself of loosing frames by having >1 thread. In fact, what I'm doing is collecting frames from the camera with one thread now, putting them to memory (let say a vector of frames) and use another different thread to save them from memory to disk. This way I randomly loose frames, since there is a moment in which something blocks the reading new frames from the camera. No way up to now to now what it is.

xlz, 2016-05-29:
In your case you were running OpenCL through Nvidia driver, which is bad. With Nvidia driver, CUDA > OpenGL > OpenCL. We know this https://github.com/OpenKinect/libfreenect2/wiki/Performance

In my experience the lag spikes are usually caused by memory allocation in the graphics stack. Using CUDA would get you a much better latency consistency.

closed 2016-05-29 by xlz