OpenKinect/libfreenect2 #659: No device found with with NiView2

twmht, 2016-05-19:
Trying to `sudo apt-get install openni2-utils && sudo make install-openni2 && NiViewer2`

but it shows

```
openDevice failed:
	DeviceOpen using default: no devices found

Press any key to continue . . .
```

However, `./bin/Protonect` works well.

What's the possible cause?

thank you

hanyazou, 2016-05-21:
please refer #632

xlz, 2016-05-29:
True as above, it's probably because you installed OpenNI not from apt-get but from a binary and your OpenNI.ini does not point to the system OpenNI library path.

closed 2016-05-29 by xlz

birgersp, 2016-06-14:
Having the exact same issue, and I don't quite understand how to actually fix this... Tried modifying `/etc/openni2/OpenNI.ini` by adding `Repository=/usr/lib/OpenNI2/Drivers/`. Is that it? I still get the same error...

doom-black, 2018-02-19:
Make symlink /usr/lib/OpenNI2/Drivers/libDummyDevice.so.0 for /usr/lib/OpenNI2/Drivers/libFreenectDriver.so.0.6.0.

doom-black, 2018-02-19:
You should make symlink /usr/lib/OpenNI2/Drivers/libDummyDevice.so.0 for freenect driver (libFreenectDriver.so.0.6.0 or something like that). 
[https://github.com/OpenKinect/libfreenect2/issues/639](url)