OpenKinect/libfreenect2 #661: Time delay and low frame rates on Jetson TK1 (ubuntu 14.04) and KinectV2 system

Senthilhariharan, 2016-05-25:
I have interfaced a KinectV2 with Jetson TK1 running on ubuntu 14.04. Protonect code seems to work. I am able to see the RGB and depth images but the frame rates are very low. The live video feed is with a huge delay. 
What have I setup wrong ?

Version, Platform, and Hardware Bug Found:

      ubuntu@tegra-ubuntu:~/libfreenect2/build/bin$ ./Protonect 
      Version: 0.2.0
      Environment variables: LOGFILE=<protonect.log>
      Usage: ./Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
              [-noviewer] [-norgb | -nodepth] [-help] [-version]
              [-frames <number of frames to process>]
      To pause and unpause: pkill -USR1 Protonect
      [Info] [Freenect2Impl] enumerating devices...
      [Info] [Freenect2Impl] 10 usb devices connected
      [Info] [Freenect2Impl] found valid Kinect v2 @2:11 with serial 500957242442
      [Info] [Freenect2Impl] found 1 devices
      [Info] [Freenect2DeviceImpl] opening...
      [Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
      [Info] [Freenect2DeviceImpl] opened
      [Info] [Freenect2DeviceImpl] starting...
      [Debug] [Freenect2DeviceImpl] status 0x090000: 8704
      [Debug] [Freenect2DeviceImpl] status 0x090000: 9729
      [Debug] [Freenect2DeviceImpl] status 0x090000: 9731
      [Info] [Freenect2DeviceImpl] submitting rgb transfers...
      [Info] [Freenect2DeviceImpl] submitting depth transfers...
      [Debug] [DepthPacketStreamParser] not all subsequences received 0
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Debug] [DepthPacketStreamParser] not all subsequences received 1016
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Info] [Freenect2DeviceImpl] started
      device serial: 500957242442
      device firmware: 4.0.3911.0
      [Debug] [DepthPacketStreamParser] subpacket too large
      [Debug] [DepthPacketStreamParser] not all subsequences received 511
      [Debug] [DepthPacketStreamParser] not all subsequences received 256
      [Debug] [DepthPacketStreamParser] not all subsequences received 448
      [Debug] [DepthPacketStreamParser] image data too short!
      [Debug] [DepthPacketStreamParser] not all subsequences received 4
      [Debug] [DepthPacketStreamParser] not all subsequences received 1020
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [RgbPacketStreamParser] skipping rgb packet!
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [RgbPacketStreamParser] skipping rgb packet!
      [Info] [DepthPacketStreamParser] 4 packets were lost
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [RgbPacketStreamParser] skipping rgb packet!
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
      [Debug] [RgbPacketStreamParser] skipping rgb packet!
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [RgbPacketStreamParser] skipping rgb packet!
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Info] [DepthPacketStreamParser] 8 packets were lost
      [Debug] [DepthPacketStreamParser] skipping depth packet
      [Info] [Freenect2DeviceImpl] stopping...
      [Info] [Freenect2DeviceImpl] canceling rgb transfers...
      [Info] [Freenect2DeviceImpl] canceling depth transfers...
      [Info] [Freenect2DeviceImpl] stopped
      [Info] [Freenect2DeviceImpl] closing...
      [Info] [Freenect2DeviceImpl] releasing usb interfaces...
      [Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
      [Info] [Freenect2DeviceImpl] closing usb device...
      [Info] [Freenect2DeviceImpl] closed
      [Info] [Freenect2DeviceImpl] closing...
      [Info] [Freenect2DeviceImpl] already closed, doing nothing
      ubuntu@tegra-ubuntu:~/libfreenect2/build/bin$ 

I am also unsure if the kinect is interfaced with a USB 3.0 Port .
the following is what i get when i run lsusb on terminal

      ubuntu@tegra-ubuntu:~$ lsusb
      Bus 002 Device 012: ID 045e:02d8 Microsoft Corp. 
      Bus 002 Device 003: ID 045e:02d9 Microsoft Corp. 
      Bus 002 Device 002: ID 05e3:0617 Genesys Logic, Inc. 
      Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
      Bus 001 Device 005: ID 045e:02d9 Microsoft Corp. 
      Bus 001 Device 004: ID 046d:c077 Logitech, Inc. 
      Bus 001 Device 003: ID 046d:c31d Logitech, Inc. Media Keyboard K200
      Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
      Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
      Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
      ubuntu@tegra-ubuntu:~$ lsusb -t
      /:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=tegra-ehci/1p, 480M
      /:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/2p, 5000M
          |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
              |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/1p, 5000M
                  |__ Port 1: Dev 12, If 0, Class=Vendor Specific Class, Driver=, 5000M
                  |__ Port 1: Dev 12, If 1, Class=Vendor Specific Class, Driver=, 5000M
                  |__ Port 1: Dev 12, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
                  |__ Port 1: Dev 12, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
      /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/6p, 480M
          |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
              |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
              |__ Port 1: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
               |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
              |__ Port 3: Dev 5, If 0, Class=Hub, Driver=hub/1p, 480M
     

Any information on this would be really helpful. Thanks in advance.

xlz, 2016-05-26:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#jetson-tk1-issues

Senthilhariharan, 2016-05-26:
Thank you. But the troubleshooting tells me to enable the USB 3.0 port in the config file. In my case it is already enabled as seen in the second output block. What can cause this delay in the video playback ?

xlz, 2016-05-29:
The viewer is not an essential part of the library so if -noviewer works fine, it is fine.

closed 2016-05-29 by xlz