OpenKinect/libfreenect2 #663: Protonect failure with USB 3.0 active extension cables

Hooges, 2016-06-01:
Hello all,

I have two ros setups which are currently running the libfreenect2 driver. The primary difference between the two setups is in the cpus. Both setups correctly run Protonect when the kinect is plugged directly into their respective usb 3.0 ports.

Recently I have been using 2x10m usb 3.0 active extension cables with my kinect. The setup with the faster cpu will still run the kinect just fine. However, the slower cpu will not run protonect correctly. The best I can tell, everything in Protonect looks like it should be running just fine, but the viewer never loads any data. Any ideas what could be causing this issue?

Thanks,

Jonathan

xlz, 2016-06-05:
"Active" probably means there is some intermediate embedded chips doing data relay. It is likely this kind of systems are not certified to transferring the amount of isochronous data for Kinect.

Hooges, 2016-06-05:
That's what I was thinking at first, but then I tried the faster setup and it worked fine with 2 cords. That makes me think it isn't the cords.

xlz, 2016-06-06:
The viewer is not important. If `Protonect -noviewer` outputs framerate information it is ok.

prarobo, 2016-10-14:
I am seeing problems with active-usb cables (1x10m usb 3.0 active extension). My motherboard has 3 usb3 roothubs. I also have a usb3 host controller plugged into PCIe expansion slot. I am using Ubuntu 14.04, Kernel 4.4.

To begin with, I am able to connect a kinect directly to ports in each of the 3 usb roothubs and I am able to get everything working. I start seeing some issues when I use active usb cables.

I have 4 different 1x10m usb 3.0 active extension cables (all of them are identical). All 4 of them work when they are plugged into the usb3 host controller plugged to PCIe expansion slot. When I plug these cables into the other usb3 ports associated with different usb3 roothubs, they sometimes work and sometimes don't.

One case I am seeing is the kinect gets detected but no data comes out. kern.log output looks clean

```
 [103895.993272] usb 6-3.1.1.1: reset SuperSpeed USB device number 93 using xhci_hcd
>[103896.099805] xhci_hcd 0000:05:00.0: WARN Event TRB for slot 8 ep 2 with no TDs queued?
```
The protonect output is this

```
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.9365ms -> ~59.0441Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.8095ms -> ~63.2531Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.1958ms -> ~58.1538Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.0909ms -> ~58.5107Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.8379ms -> ~63.1396Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.6029ms -> ~64.0908Hz
.....
.....
```

In some other cases, the kinect doesn't even get detected properly. It looks like only the NUI Sensor Adapter hub is detected but none of the devices connected to it are getting detected. The part of the usb-devices output is:

```
T:  Bus=03 Lev=03 Prnt=125 Port=00 Cnt=01 Dev#=126 Spd=480 MxCh= 1
D:  Ver= 2.10 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=045e ProdID=02d9 Rev=00.70
S:  Manufacturer=Microsoft Corporation  
S:  Product=NuiSensor Adaptor      
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
```
lsusb output

```
Bus 002 Device 002: ID 8087:8002 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:800a Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 003 Device 114: ID 413c:2113 Dell Computer Corp. 
Bus 003 Device 126: ID 045e:02d9 Microsoft Corp. 
Bus 003 Device 125: ID 2109:2813 VIA Labs, Inc. 
Bus 003 Device 124: ID 2109:2813 VIA Labs, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Can I change any low-level usb parameters and do some tests. Any help would be appreciated. Thanks,

prarobo, 2016-10-14:
I figured out my problem. I am posting here for others who see such issues.

The problem was that the usb ports on the motherboard were not able to support the power draw of the active usb cable (the usb controller on the pcie expansion slot worked as it had a dedicated power line). I had previously plugged in a separate external power supply on the other end of the active usb cable but the end that is connected to the pc also required a powered usb-hub to fix my connection problem.

On top of this one of my 4 cables was bad. Hope this helps someone else.

xlz, 2016-12-05:
I consider this resolved.

closed 2016-12-05 by xlz