OpenKinect/libfreenect2 #666: Protonect.exe: failed to create usb context: LIBUSB_ERROR_NOT_FOUND Entity not found

HollaDieWaldFee0815, 2016-06-13:
Hello,
i followed the installation instructions and used Zadig to install the libusbk driver. And build the libfreenect2 library succesfully.

Using: Libfreenect2, Win7, Intel® Q85

When i try to run the Protonect.exe i get this error message:
[image]
	
In the device manager i can see the Xbox NUI Sensor(Composite Parent) listed under "libusbk USB devices". But i had to install some usb 3.0 intel drivers to see the device listed, before that i didn´t see anything.

The kinect v2 sensor works flawlessly on another computer. So i don´t think its an issue with the sensor.

Thanks in advance for your help.

xlz, 2016-06-14:
Is USB 3.0 supported on this computer?

HollaDieWaldFee0815, 2016-06-14:
It is supported on this computer. I got the same pc with a linux os, and the sensor is working fine there with libfreenect2.

HollaDieWaldFee0815, 2016-06-16:
I found the solution. I used libusbx from josh blake (see : https://github.com/wjchen84/libfreenect2 )
and voila its working fine!

closed 2016-12-05 by xlz