OpenKinect/libfreenect2 #667: RFC: Features for 0.3?

xlz, 2016-06-14:
What features do you want to be included in 0.3?

* USB zerocopy based performance improvement
* Better platform feature detection and configuration (compile with full features, detect and load available decoders at runtime)
* Gray output format of JPEG decoders
* Firmware upload. The protocol has been reverse engineered, using Windows to do this is still easy.
* Example of multiple Kinects.
* Example utility of dumping image frames.
* API for on-demand processing.
* Bindings for C, Python, Java, etc.

Others?

chengzijiazu, 2016-06-14:
Hi,Xianghu
I hope that the feature of example of multiple Kinects and skeleton tracking can be included in 0.3.These features are very useful. Thank you  for your hard working.Graet

BrannonKing, 2016-06-24:
pre-built packages for Fedora 24, Ubuntu 16.04

jasoncatchpole, 2016-07-01:
- USB zerocopy based performance improvement
- Example of multiple Kinects
- Firmware upload (but only if it is easy). There aren't many updates to the firmware from Microsoft to warrant too much work on this?
- Option for un-mirroring depth & RGB images as per [this issue](https://github.com/OpenKinect/libfreenect2/issues/172)

Keep up the awesome work!

hanshammel1337, 2016-12-20:
- Add a option to mirror the depth image