OpenKinect/libfreenect2 #668: Os X Install instructions don't work for OpenNI

tetmin, 2016-06-21:
I've followed the install instructions for OS X to the letter, everything is working fine except for the last part:

Test OpenNI2. make install-openni2 (may need sudo), then run NiViewer

NiViewer does not exist anywhere on my system so i'm not able to run it.

Thanks

cupofnestor, 2016-06-21:
I found an executable in `/usr/local/share/openni2/tools` however I do not see a new window appear when I run it, just the stdout from the process.

tetmin, 2016-06-21:
Thanks, just ran it and also only seeing the text output.

tetmin, 2016-06-21:
Also is there any clear information anywhere on how to get the skeleton tracking working with NiTE on OS X or Linux?

panzer27, 2016-08-04:
Hey guys, any update ? can't open NiViewer.

orgicus, 2016-08-28:
Same issue: OpenNI2 libfreenect driver doesn't seem to be picked up.

Here's the output from NiViewer:
```bash
openDevice failed:
	DeviceOpen using default: no devices found

Press any key to continue . . .
```

I've installed OpenNI2 via homebrew as instructed in the README:

```/usr/local/Cellar/openni2/2.2.0.33/share/openni2/tools/NiViewer```

and I can see the driver installed as expecetd:

```ls /usr/local/Cellar/openni2/2.2.0.33/share/openni2/tools/OpenNI2/Drivers/
PS1080.ini			PSLink.ini			libOniFile.dylib		libPS1080.dylib			libPSLink.dylib			libfreenect2-openni2.0.dylib	libfreenect2-openni2.dylib
```

The only thing that looked odd during the build process was this warning:
```bash
 CMake Warning (dev):
   Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
   --help-policy CMP0042" for policy details.  Use the cmake_policy command to
   set the policy and suppress this warning.

   MACOSX_RPATH is not specified for the following targets:

    freenect2
    freenect2-openni2

 This warning is for project developers.  Use -Wno-dev to suppress it.
```
Is this something to worry about ?

What's the simplest way to use NITE Skeleton tracking with libfreenect2 on OSX ?

kryzhan, 2017-02-07:
Hi guys, any news? I'm trying to show Window of NiViewer, no luck.

Volodymyrs-MBP:~ dyno$ /usr/local/Cellar/openni2/2.2.0.33/share/openni2/tools/NiViewer ; exit;
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 4 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:8 with serial 005145662047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 4 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:8 with serial 005145662047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 4*128*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Info] [Freenect2DeviceImpl] started
**2017-02-08 00:01:04.107 NiViewer[4809:69038] GLUT Warning: glutInit being called a second time.**
[Info] [DepthPacketStreamParser] 9 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 12.696ms -> ~78.7648Hz

Cannot find any solution.

hanyazou, 2017-02-08:
Did you try another packet processors? I'm afraid that the OpenGLDepthPacketProcessor is conflicting with the GLUT.

kryzhan, 2017-02-08:
I am not completely understood what do you mean. I found this issue - https://github.com/OpenKinect/libfreenect2/issues/386 And tried to change code somehow similar:
`  // setup context
  glfwDefaultWindowHints();
  glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3);
#ifdef __APPLE__
  glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 3);
  glfwWindowHint(GLFW_OPENGL_FORWARD_COMPAT, GL_TRUE);
  glfwWindowHint(GLFW_OPENGL_PROFILE, GLFW_OPENGL_CORE_PROFILE);
#else
  glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 3);
  glfwWindowHint(GLFW_OPENGL_PROFILE, GLFW_OPENGL_CORE_PROFILE);
#endif
  glfwWindowHint(GLFW_VISIBLE, debug ? GL_TRUE : GL_FALSE);

  GLFWwindow* window = glfwCreateWindow(1024, 848, "OpenGLDepthPacketProcessor", 0, parent_window);
`
But result is the same.

closed 2017-04-04 by xlz

laihaotao, 2018-07-13:
any update here? same issue, cannot see the GUI of the viewer?

chilippso, 2021-06-28:
I am (still today) experiencing the same issue. No GUI for NiViewer.

`2021-06-28 21:52:11.597 NiViewer[13246:577628] GLUT Warning: glutInit being called a second time.`