OpenKinect/libfreenect2 #669: Error, when I install OpenGL

QinZiwen, 2016-06-22:
I had use a ubuntu14.04.2 computer, when I run "sudo apt-get install libgl1-mesa-dri-lts-vivid", some errors show me as following:

turtlebot@turtlebot-HP-ZBook-14-G2:~/libfreenect2/build$ sudo apt-get install libgl1-mesa-dri-lts-vivid
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 unity-control-center : Depends: libcheese-gtk23 (>= 3.4.0) but it is not going to be installed
                        Depends: libcheese7 (>= 3.0.1) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

But I have aready install libcheese-gtk23 and  libcheese7. I see that the author said "If the last command conflicts with other packages, don't do it.", I ignore it and go on. running "./bin/Protonect", nothing show me.

BrannonKing, 2016-06-24:
What are your graphics cards' models? What drivers are you using for them -- native or open source? What desktop face are you running? Is it Unity?

QinZiwen, 2016-06-26:
My graphics is Intel HD graphics 5500; desktop face is Unity

xlz, 2016-06-30:
The doc says:

> (If the last command conflicts with other packages, don't do it.)

closed 2016-06-30 by xlz

QinZiwen, 2016-07-01:
yes, I do. thank you @xlz