OpenKinect/libfreenect2 #671: Cannot correctly set up the OpenCL stack

cforntu, 2016-06-23:
Ubuntu 14.04
After I installed the opencl, I want to verify it using clinfo, but the output is 
 X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
X server found. dri2 connection failed! 
/dev/dri/card0 not authenticated
beignet-opencl-icd: no supported GPU found, this is probably the wrong opencl-icd package for this hardware
 and I reinstall the opencl but the error is still there. Could any give me any suggestion

BrannonKing, 2016-06-24:
What are your graphics cards' models? What drivers are you using for them -- native or open source?

cforntu, 2016-06-27:
My graphic card model is
   product: GM200 [GeForce GTX 980 Ti]
And this is my driver information:
  
[image]
  Could you please help me to solve the problem

xlz, 2016-06-30:
Use CUDA.

closed 2016-06-30 by xlz