OpenKinect/libfreenect2 #672: Getting libusb error on running Protonect...

nivtri, 2016-06-24:
Around 2 weeks back I was able to view the point cloud from my Kinect v2 sensor using `./Protonect`. Everything was working perfectly fine. After that I could not use my system for some time and now after getting back my system when I executed Protonect, I got following error:
**[Error] [Freenect2Impl] Your libusb does not support large iso buffer!
no device connected!**
Complete output is:
`Version: 0.2.0`
`Environment variables: LOGFILE=<protonect.log>`
`Usage: ./Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]`
       ` [-noviewer] [-norgb | -nodepth] [-help] [-version]`
      `  [-frames <number of frames to process>]`
`To pause and unpause: pkill -USR1 Protonect`
`[Error] [Freenect2Impl] Your libusb does not support large iso buffer!`
`no device connected!`

I have no idea what got changed. I also followed instructions given at [troubleshooting page](https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting), but couldn't get rid of that error. Please help me out to solve this.

- I am using USB 3.0 port and Ubuntu 14.04 Version. Output of `uname -a`:
`Linux roboticlab-HP-ProBook-450-G2 3.19.0-56-generic #62~14.04.1-Ubuntu SMP Fri Mar 11 11:03:15 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
`

- Output of `ldd ./Protonect | grep libusb`:
`libusb-1.0.so.0 => /lib/libusb-1.0.so.0 (0x00007f255df4e000)`

- Output of `dpkg -l libusb-1.0*`:
`Desired=Unknown/Install/Remove/Purge/Hold`
`| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend`
`|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)`
`||/ Name           Version      Architecture Description`
`+++-==============-============-============-=================================`
`ii  libusb-1.0-0:a 2:1.0.20-1   amd64        userspace USB programming library`
`ii  libusb-1.0-0-d 2:1.0.20-1   amd64        userspace USB programming library`
`ii  libusb-1.0-doc 2:1.0.17-1ub all          documentation for userspace USB p`

Thanks!

Abbyls, 2016-08-25:
Hi, i'm confused with the same problem! Have you salved it ? Can you tell me how to deal with it?

pranali3215, 2016-11-22:
Hi,
Can you tell me the output of 
 > pkg-config --debug libusb-1.0

xlz, 2016-12-05:
You have some libusb older than 1.0.20 somewhere in your system.  Remove it.

NOTABUG.

closed 2016-12-05 by xlz