OpenKinect/libfreenect2 #674: OpenCL failed to build program (on AMD Tahiti)

BrannonKing, 2016-06-24:
I'm attempting to use the OpenCL handler on a fancy MacPro running Fedora 24. It has two AMD GPUs: Radeon HD 7870 (Tahiti XL). I tried the available Beignet package and I compiled their latest head myself. Both give a similar error. The output from Protonect looks like this:

```
selected device: AMD TAHITI (DRM 2.43.0, LLVM 3.8.0) (GPU)[AMD]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Error] [OpenCLDepthPacketProcessorImpl] program.build(options.c_str()): -43
[Error] [OpenCLDepthPacketProcessorImpl] failed to build program: -43
[Error] [OpenCLDepthPacketProcessorImpl] Build Status: -2
...
[Error] [OpenCLDepthPacketProcessorImpl] Build Log:	
[Error] [OpenCLDepthPacketProcessorImpl] OpenCLDepthPacketProcessor is not initialized!
```

Does that build error mean anything? Is there some other more verbose OpenCL build error somewhere else? I've attached my clinfo ouput: 
[clinfo.txt](https://github.com/OpenKinect/libfreenect2/files/332242/clinfo.txt)

xlz, 2016-10-09:
-43 is CL_INVALID_BUILD_OPTIONS.

We are using some build options here https://github.com/OpenKinect/libfreenect2/blob/master/src/opencl_depth_packet_processor.cpp#L340

Probably one of them is not supported by your driver? You can try removing one of the options and see if that works.

xlz, 2017-04-04:
No update so I assume it's a driver issue.

closed 2017-04-04 by xlz