OpenKinect/libfreenect2 #675: I met a problem when I create new project on visual studio

TaroTokui, 2016-06-29:
I met a problem when I create a new project on visual studio

I installed latest libfreenect2 on my windows10(64bit) with UsbDk driver.
And I could build & run sample project(Protonect) on visual studio 2015.

When I create a new project on visual studio with libfreenect2, it can be build.
But it get "Read Access violation error" at geDefaultDeviceSerialNumber() on debug mode. (Although enumerateDevices() returns 1.)

Here are my project settings.

- additional include directories

 (lifbreenect2 directory)¥build¥install¥include

- additional library directories

 (lifbreenect2 directory)¥build¥install¥lib
 (lifbreenect2 directory)¥depends¥libusb¥MS64¥dll
 (lifbreenect2 directory)¥depends¥glfw¥lib-vc2015
 (lifbreenect2 directory)¥depends¥libjpeg-turbo64¥lib

- additional dependencies

 freenect2.lib
 libusb-usbdk-1.0.lib
 turbojpeg.lib
 glfw3dll.lib

Do you have any idea?

Thanks.

xlz, 2016-06-30:
Try libusbK instead of UsbDk?

TaroTokui, 2016-07-01:
@xlz Thank you for your advice. 
I've tried libusbK, but I got same error.

TaroTokui, 2016-07-01:
I found the solution of this problem.
There are some mistakes at “additional library directories" and "additional dependencies."

Correct settings are here.

for debug mode

- additional library directories
(lifbreenect2 directory)¥build¥lib¥Debug
(lifbreenect2 directory)¥depends¥libusb¥MS64¥dll
(lifbreenect2 directory)¥depends¥glfw¥lib-vc2015
(lifbreenect2 directory)¥depends¥libjpeg-turbo64¥lib

- additional dependencies
freenect2d.lib
libusb-usbdk-1.0.lib
turbojpeg.lib
glfw3dll.lib

for release mode

- additional library directories
(lifbreenect2 directory)¥build¥lib¥Release
(lifbreenect2 directory)¥depends¥libusb¥MS64¥dll
(lifbreenect2 directory)¥depends¥glfw¥lib-vc2015
(lifbreenect2 directory)¥depends¥libjpeg-turbo64¥lib

- additional dependencies
freenect2.lib
libusb-usbdk-1.0.lib
turbojpeg.lib
glfw3dll.lib

closed 2016-07-01 by TaroTokui

adri1992, 2017-11-02:
Hi guys!

I've the same problem. In debug mode, it has been solved with @TaroTokui settings, but in Release mode the problem has not disappear... =( I'm going crazy.
I've change the additional library directories to the Release path .

Any idea of what can I try?