OpenKinect/libfreenect2 #676: Raspberry Pi 3 Kinect v2 connection problem - no BOS descriptor

simontoao, 2016-06-29:
Overview Description:

I have Raspberry Pi 3 and Kinect v2 connected via Kinect Adapter. I have installed libfreenect2 without errors and when I try to run Protonect I get this error message:

[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @1:13 with serial 008335244747
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBU$
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @1:13
failure opening device!

This is what dmesg reutrns after reconnecting device and running Protonect:

[11986.415942] usb 1-1.4: USB disconnect, device number 12
[11986.415966] usb 1-1.4.1: USB disconnect, device number 13
[11991.256531] usb 1-1.4: new high-speed USB device number 14 using dwc_otg
[11991.358959] usb 1-1.4: New USB device found, idVendor=045e, idProduct=02d9
[11991.358981] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[11991.358994] usb 1-1.4: Product: NuiSensor Adaptor
[11991.359007] usb 1-1.4: Manufacturer: Microsoft Corporation
[11991.359970] hub 1-1.4:1.0: USB hub found
[11991.360150] hub 1-1.4:1.0: 1 port detected
[11993.036613] usb 1-1.4.1: new high-speed USB device number 15 using dwc_otg
**[11993.146968] usb 1-1.4.1: unable to get BOS descriptor**
[11993.147978] usb 1-1.4.1: New USB device found, idVendor=045e, idProduct=02c4
[11993.147994] usb 1-1.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[11993.148007] usb 1-1.4.1: Product: Xbox NUI Sensor
[11993.148019] usb 1-1.4.1: Manufacturer: Microsoft
[11993.148032] usb 1-1.4.1: SerialNumber: 008335244747
[11995.276682] usb 1-1.4.1: reset high-speed USB device number 15 using dwc_otg
**[11995.386843] usb 1-1.4.1: unable to get BOS descriptor**

It says that there is a problem with getting BOS descriptor. How to fix this problem? Is it a matter of dwc_otg configuration? 

Version, Platform, and Hardware:

Linux raspberrypi 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 14, If 0, Class=Hub, Driver=hub/1p, 480M
            |__ Port 1: Dev 15, If 0, Class=Vendor Specific Class, Driver=, 480M
            |__ Port 1: Dev 15, If 1, Class=Audio, Driver=snd-usb-audio, 480M
            |__ Port 1: Dev 15, If 2, Class=Audio, Driver=snd-usb-audio, 480M

xlz, 2016-06-30:
Does RPi3 support USB 3?

simontoao, 2016-07-01:
No, there are only usb 2.0 ports, so that must be the reason it is not working properly. I have assumed that they have upgraded it since last version, but they didn't. Thank you.