OpenKinect/libfreenect2 #677: Mutex Lock Error in waitForNewFrame

nmagerko, 2016-06-29:
### Overview

Calling `waitForNewFrame` causes the following error (read access violation): 
```
_Mtx_internal_imp_t::_get_cs(...)-> was nullptr.
```

Version: libfreenect2 v0.2
Platform: Windows 10 Home, x64
Hardware: Kinect for Windows v2, USB 3.0

### Reproduce

1. Enumerate available devices. Open the device at index 0 and store the resulting pointer. Initialize a frame listener (`SyncMultiFrameListener`) using IR, Depth, and Color flags. Set the color and frame listeners for the device at index 0.
2. Start the device at index 0 via the `start()` method
3. Attempt to collect at least 2 separate frames using the previously-declared `SyncMultiFrameListener` (call `waitForNewFrame`, passing in a pointer to a `FrameMap`

### Actual Results

Process enters `frame_listener_impl.cpp` to wait before collecting the first frame. Execution starts at line 134, `libfreenect2::unique_lock l(impl_->mutex_);`. An error is ultimately caught in line 99 of `mutex.c`, `mtx->_get_cs()->lock();`.

### Expected Results

The process waits indefinitely for a frame to be captured by the Kinect, and then allows the process to continue.

### Reproducibility

An error is caught without fail. However, the value of `_get_cs()` seems to change during different runs. Sometimes, the value is 0xFFFFFFFFFFFFFFFF while other times it is 0x40. Most commonly, though, the value is nullptr. 

### Program Output
```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 17 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:10 with serial xxxxxxxxx542
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
```

xlz, 2016-06-30:
Good report. Though I don't have time to look at it in a while. If you could figure out a fix it'd be great.

nmagerko, 2016-06-30:
@xlz Here's what seems to be the problem:

- the [SyncMultiFrameListener constructor](https://github.com/OpenKinect/libfreenect2/blob/master/src/frame_listener_impl.cpp#L89) initializes `_impl` with a pointer to a SyncMultiFrameListenerImpl in dynamic memory
- the program I am writing implicitly used `operator=` to assign the content of a newly-created `SyncMultiFrameListener` to a variable holding an existing instance
- since no `operator=` is defined, a shallow copy is performed. The existing instance receives the newly-created listener's mutex. However, the newly-created listener subsequently went out of scope in my code, causing the copied mutex to be deleted in memory by the deconstructor.
- the existing listener (now holding a bad reference to its mutex) tries to unlock the mutex, which causes a access violation 

Bottom line: this class needs to implement the Rule of Three. It has a deconstructor, but lacks an implementation of its copy assignment operator (and copy constructor for that matter).

nmagerko, 2016-06-30:
Stepping through the rest of my code, it seems like other classes might have this issue too. For example, the `Registration` class is using dynamic memory (and has a deconstructor), but lacks a copy constructor and `operator=` definition.

xlz, 2016-12-05:
Your comments are on the point. There are (only) three classes `Registration`, `SyncMultiFrameListener`, and `Freenect2` that use the pimpl pattern. If I were to write it now in C++11 the pimpl would be wrapped in unique_ptr and the issue with implicit copy/assignment constructor would be gone.

But that is some C++98 historical baggage API which isn't worth converting to C++11 for right now. So I'll just disable the constructors for the three classes for now.

xlz, 2016-12-05:
I created #765 to fix this. I'll give it some time for review before merging it.

commit d63b15f4 (2016-12-13): Disable default copy and assignment constructors

To prevent locks from being copied.

Close #677.
closed 2016-12-05 by xlz