OpenKinect/libfreenect2 #681: [Question] Depth processor stuck when preserve depth frames rather than deleting them instantly

dujiulun, 2016-07-06:
**Overview Description:**
If I preserve a depth frame and delete it later, the depth processor is stuck ("skipping depth packet") until the last frame is deleted.

In my application, I copied and modified the codes from `Protonect` example and `SyncMultiFrameListener`. I use `Protonect`'s main thread to `waitForNewFrame()`, and preserve the color and depth frames for batch delete once per second. Then I modified `SyncMultiFrameListener::release()` so it doesn't call `Frame`'s destructor. After that, I can only get one group of frames per second, and no more until I delete these frames.

Then I did more testing, and realized that only depth processor is stuck (I used CUDA to process depth frames). If I use a custom listener, I can get 30fps color, but only 1 depth frame unless I delete the previous one.

I dug through the code but couldn't find where to change to fix this. Could you please help?

**Version, Platform, and Hardware Bug Found:**
```
1. `git log -1 --oneline`
83f88b4 Update author list for 0.2 release
2. `uname -a`
Linux tegra-ubuntu 3.10.40-gdacac96 #1 SMP PREEMPT Thu Jun 25 15:25:11 PDT 2015 armv7l armv7l armv7l GNU/Linux
3. `lsusb -t`
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/2p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
4. `lspci -nn`
00:00.0 PCI bridge [0604]: NVIDIA Corporation TegraK1 PCIe x1 Bridge [10de:0e13] (rev a1)
01:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 0c)
```

**Steps to Reproduce:**

1. Use CUDA depth processor. After getting a depth `Frame`, don't destruct it.
2. Wait for more frames.
3. 

**Actual Results:**
No more depth frames will be received. The debug log shows "skipping depth packet" lines.

**Expected Results:**
More depth frames should come.

**Reproducibility:**

**Additional Information:**

xlz, 2016-07-07:
This is expected behavior.

https://github.com/OpenKinect/libfreenect2/commit/709a5b986a9c32fe1f368e18ea0b89696d5ca4c2: 
> now SyncMultiFrameListener returns
false and does not save any frames before the user releases
the frame map.

You must release the frame map to receive new frames. If you do not release the frame map within 33ms you will lose frames. Especially in the case of Tegra, the frame contains pointers to opaque Nvidia memory which is not allocated by us and not possible to be freed by us therefore we need to recycle it immediately.

What you can do is finish whatever image processing with the frame within 33ms or memcpy it to somewhere else and release it.

closed 2016-07-07 by xlz

dujiulun, 2016-07-08:
Hi @xlz ,

Thanks for the response. I understand how `SyncMultiFrameListener` works, so I modified it such that when I call `release()` the frames aren't actually deleted.

It is the second part that interests me. When I use my custom frame listener, I also have to delete the frame to free the depth processor. You explained that the "Nvidia memory" wasn't allocated by libfreenect2, but I found a `CudaAllocator` class in `cuda_depth_packet_processor.cu`. Is there any way to modify it to allocate more memory and save more frames?

I tried the memcpy method too, but it causes lag in both RGB and depth even if I only copy the depth frames, because the Tegra has such low processing power.

Thanks!

xlz, 2016-07-08:
There is no point in using more than double buffer if your code reads data slower than the data comes. Same for saving.

dujiulun, 2016-07-08:
@xlz , but if I can save one or several frames, I can use a separate thread (hopefully a separate CPU core) to process the previous frames while receive the next ones.

My point is, in the current implementation, each depth frame takes ~20ms to be processed, leaving me ~13ms (compared to a full ~33ms with a separate thread) to write it to disk. If I don't call the frame destructor in 13ms, the following depth packets and frames will be dropped.

Could you point to me where I can make the change so that the depth packets will not be "skipped"?

xlz, 2016-07-08:
Why do you think a separate thread would make things better?

The current implementation is chained double buffers. It gives you 33ms, not 13ms.

I cannot. The only way to do this is rewrite everything. You can try, but I will not support it.

dujiulun, 2016-07-08:
@xlz , this clears it up. I misunderstood the code and thought it was only one buffer. Thank you for the help.