OpenKinect/libfreenect2 #683: libva error: va_getDriverName() failed with unknown libva error,driver_name=(null) [Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error

muwangchuzhong, 2016-07-10:
when I run "./bin/Protonect",this error message repeatedly pops up and protonect is black:
lyz@lyz-All-Series:~/libfreenect2/build$ ./bin/Protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:15 with serial 021705643547
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9759
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 021705643547
device firmware: 4.0.3911.0
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.0469ms -> ~58.6618Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.3546ms -> ~57.6215Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 12.9928ms -> ~76.9655Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.8778ms -> ~62.9812Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.2314ms -> ~58.0335Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.2805ms -> ~57.8689Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.014ms -> ~55.5125Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.9203ms -> ~55.8027Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 15.5403ms -> ~64.3488Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.2667ms -> ~57.9148Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.8039ms -> ~56.1674Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.9038ms -> ~55.8542Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.8288ms -> ~59.4218Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.7956ms -> ~59.5394Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.4326ms -> ~57.3637Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.5105ms -> ~60.5675Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.2362ms -> ~58.0173Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.6972ms -> ~56.5061Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.0384ms -> ~58.691Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.1253ms -> ~55.1715Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.4528ms -> ~57.2974Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.7238ms -> ~56.4214Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.9738ms -> ~58.9143Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.826ms -> ~56.0978Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.9807ms -> ~55.6151Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 16.9167ms -> ~59.1132Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 17.9422ms -> ~55.7347Hz
I would like to seek everyone's help，thanks

xlz, 2016-07-18:
It only appears once in your log so what is "repeatedly" popping up?

This message tells you libva does not work on your machine somehow and libfreenect2 falls back to turbojpeg.

Dyddye, 2016-08-31:
I just tested with Ubuntu 14.04 and have the same issue. As root (`sudo ./bin/Protonect gl`) it succeedes. Looks like it has something to to with permissions.

`./bin/Protonect gl`
> libva info: VA-API version 0.39.0
> libva info: va_getDriverName() returns 1
> libva error: va_getDriverName() failed with operation failed,driver_name=i965
> [Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): operation failed

`sudo ./bin/Protonect gl`
> libva info: VA-API version 0.39.0
> libva info: va_getDriverName() returns 0
> libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
> libva info: Found init function __vaDriverInit_0_39
> libva info: va_openDriver() returns 0
> [Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Ivybridge Mobile - 1.7.0

xlz, 2016-09-06:
Ivybridge GPU does not have JPEG decoding.

xlz, 2016-09-06:
Oh, the above is wrong. Ivybridge is the minimum. Yes there can be some issue with /dev/dri*. Probably you should have added yourself to video group.

closed 2016-09-06 by xlz

Dyddye, 2016-09-06:
Indeed belongs /dev/dri to root:video:

```
$ ls -la /dev/dri/
total 0
drwxr-xr-x   2 root root      100 Sep  6 10:10 .
drwxr-xr-x  17 root root     4360 Sep  6 10:10 ..
crw-rw----+  1 root video 226,  0 Sep  6 10:09 card0
crw-rw----+  1 root video 226,  1 Sep  6 10:10 card1
crw-rw----   1 root video 226, 64 Sep  6 10:09 controlD64
```

I joined the video group, but still no change.

`$ sudo usermod -a -G video YOUR_NAME`

```
$ id YOUR_NAME
uid=1009(YOUR_NAME) gid=1009(YOUR_NAME) groups=1009(YOUR_NAME),4(adm),27(sudo),44(video),108(lpadmin),124(sambashare)
```

```
$ ./bin/Protonect gl
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 1
libva error: va_getDriverName() failed with operation failed,driver_name=i965
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): operation failed

```

crasylph, 2016-09-22:
I recognized that this issue has been closed, but I failed to solve the problem by trying every method mentioned above...

The worst thing is, while libva can't find the driver when i run ./bin/Protonect or sudo ./bin/Protonect gl, it works well when i run vainfo.
So what on earth may stop libva from finding the driver in libfreenect2 ?

maxcr, 2016-11-09:
Dude I'm having the same problem and its driving me nuts.

xlz, 2016-11-09:
> [Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): operation failed

I have no idea about this one.

> libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so

Remove nvidia libva wrapper or use `export LIBVA_DRIVER_NAME=i965`.

Harsharma2308, 2018-10-30:
How to solve this issue?

xlz, 2018-10-31:
Remove nvidia libva wrapper or use export LIBVA_DRIVER_NAME=i965.

lucaszanella, 2019-02-14:
After doing `export LIBVA_DRIVER_NAME=i965`

I have the following errors:

```
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed

```

medashan, 2019-07-17:
@lucaszanella Are you able to fix this problem?

kkonen, 2020-06-16:
any new insight @lucaszanella @medashan @crasylph @maxcr ?

newcanopies, 2021-04-05:
> Remove nvidia libva wrapper

hi @xlz, how to remove wrapper on: Linux 5.4.72-microsoft-standard-WSL2 #1 SMP Wed Oct 28 23:40:43 UTC 2020

where running X11 for ROS gui

TipsyTheCat, 2025-04-21:
dev gave up on this lib ages ago