OpenKinect/libfreenect2 #687: No device connected message

gracediaz, 2016-07-15:
Overview Description:
 I'm doing the steps of the webpage (github) but in the end, it is not recognizing the Kinect (it says no device detected).  My graphic card is Intel Ivybridge.  Would that be a problem?

The Kinect is connected but it isn't recognized in the program.

Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

Bus 002 Device 003: ID 0a5c:5801 Broadcom Corp. BCM5880 Secure Applications Processor with fingerprint swipe sensor
 Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
 Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
 Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
 Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
 Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
 Bus 003 Device 083: ID 045e:02bf Microsoft Corp. Kinect for Windows NUI Camera
 Bus 003 Device 082: ID 045e:02be Microsoft Corp. Kinect for Windows NUI Audio
 Bus 003 Device 081: ID 045e:02c2 Microsoft Corp. Kinect for Windows NUI Motor
 Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Could someone help me with this problem?  

Best regards,

Grace Diaz

chengzijiazu, 2016-07-16:
USB 3.0 or not?

gracediaz, 2016-07-18:
Good morning,

Yes, USB 3.0.

ibranch, 2016-10-03:
Based on the output of `lsusb` there at the end, it looks like you do have a USB 3.0 Hub on Bus 004, but your Kinect devices are plugged into the USB 2.0 Hub, as they all share Bus 003.

Perhaps double-check that the Kinect is plugged in to a USB 3.0 port, and not just that you have other USB 3.0 ports available?

xlz, 2016-12-05:
Agreed. Looks like NOTABUG.

closed 2016-12-05 by xlz

BerryOne, 2018-05-30:
I meet the same problem,it shows:
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] []
[-noviewer] [-norgb | -nodepth] [-help] [-version]
[-frames ]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 4 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

Can you help me?Thanks