OpenKinect/libfreenect2 #688: Multiple kinect

chengzijiazu, 2016-07-16:
Overview Description:
multiple kinect: I am working with the multiple kinect v2 on windows base on libfreenect2 0.2.
When I debuy the programmer,the kinects can start and programmer shut down after a while.I don't while.Could some one give me a hand? When connect with two kienct ,it can work well.

Computer		DELL Precision M6800 notebook
Cpu		         	Intel(R) Core(TM) i7-4700MQ CPU @ 2.40GHz
RAM	                	32.0GB
Graphic card			1、NVIDIA Quadro K3100M
		         	2、Intel(R) HD Graphics 4600
Disk		               1、PLEXTOR PX-128M6S+ (128GB)
			       2、HGST HTS545050A7E380 (500GB)
Mainboard		0XD1M5 (A00)
Network card		1、Intel(R) Ethernet Connection I217-LM
		         	2、Dell Wireless 1550 802.11ac

display			LGD:da02 :1920x1080 
system		       Windows 8.1 64

the code:
/*
 * This file is part of the OpenKinect Project. http://www.openkinect.org
 *
 * Copyright (c) 2011 individual OpenKinect contributors. See the CONTRIB file
 * for details.
 *
 * This code is licensed to you under the terms of the Apache License, version
 * 2.0, or, at your option, the terms of the GNU General Public License,
 * version 2.0. See the APACHE20 and GPL2 files for the text of the licenses,
 * or the following URLs:
 * http://www.apache.org/licenses/LICENSE-2.0
 * http://www.gnu.org/licenses/gpl-2.0.txt
 *
 * If you redistribute this file in source form, modified or unmodified, you
 * may:
 *   1) Leave this header intact and distribute it under the same terms,
 *      accompanying it with the APACHE20 and GPL20 files, or
 *   2) Delete the Apache 2.0 clause and accompany it with the GPL2 file, or
 *   3) Delete the GPL v2 clause and accompany it with the APACHE20 file
 * In all cases you must keep the copyright notice intact and include a copy
 * of the CONTRIB file.
 *
 * Binary distributions must follow the binary distribution requirements of
 * either License.
 */

/** @file Protonect.cpp Main application file. */

#include <iostream>
#include <cstdlib>
#include <signal.h>

/// [headers]
#include <libfreenect2/libfreenect2.hpp>
#include <libfreenect2/frame_listener_impl.h>
#include <libfreenect2/registration.h>
#include <libfreenect2/packet_pipeline.h>
#include <libfreenect2/logger.h>
/// [headers]
#ifdef EXAMPLES_WITH_OPENGL_SUPPORT
#include "viewer.h"
#endif

bool protonect_shutdown = false; ///< Whether the running application should shut down.

void sigint_handler(int s)
{
	protonect_shutdown = true;
}

bool protonect_paused = false; ///< Whether the running application should pause.
[... 481 lines cut ...]
	// TODO: restarting ir stream doesn't work!
	// TODO: bad things will happen, if frame listeners are freed before dev->stop() :(
	/// [stop]
	dev->stop();
	dev->close();
	dev1->stop();
	dev1->close();
	dev2->stop();
	dev2->close();
	/// [stop]

	delete registration;
	delete registration1;
	delete registration2;

	return 0;
}

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

djivani, 2016-07-19:
I believe you cannot run more than 2 Kinects on a single USB 3.0 controller card, which is probably what exists in your DELL Precision M6800 notebook. The solution is to use a desktop with spare PCIe slots where you can plug-in extra USB 3.0 controllers, ideally 1 for each Kinect.

chengzijiazu, 2016-07-21:
Thank you! I will try to modified the code !

kalluwa, 2016-08-16:
Hi , 
i have the same problem with U, libfreenect2 works fine with two kinects but failed with 3 kinects.
i am working at DELL precision 7510 and not sure for problems of PCIe.
have u made it workable or any suggestions?

chengzijiazu, 2016-08-16:
Not yet,but I am still working to solve this problem.

kalluwa, 2016-08-17:
after the reading the issue :
[320](https://github.com/OpenKinect/libfreenect2/issues/320)
> Yes, I can confirm that each Kinect needs its own USB-controller. Current main boards typically have one or two USB 3.0 controllers, so one needs to plug in USB 3.0 expansion card(s). In my current setup I use 4 x Transcend TS-PDU3 USB3.0 Expansion Cards and one on board controller. Running Ubuntu 14.04 LTS.

i believe that THE HARDWARE can just support 2 kinects just as djivani said, and i have to choose anothing way(change the PC) to use multiple kinects, :disappointed: 

waiting for your good news.

djivani, 2016-08-23:
I am working on a setup of 6 Kinects using 5 **Transcend USB 3.0 Dual Expansion Card PCI Express Interface Model TS-PDU3** card and the onboard USB 3.0 controller.

chengzijiazu, 2016-08-24:
Could you give me your code about the multiple kinects.（I think there must be something wrong with my code）I am starting again and working for skeleton tracking project of my postgraduate tast. @djivani

sunwtGitHub, 2016-12-08:
Hi, I am working with three number of kinect v2 on windows 10. 
I installed two Transcend USB 3.0 Expansion cards (one on a PCI-E 1x and one on a PCI-E 16x).
I am sure every devices has thier own usb3 controller.

In my program, I used three number of threads to process the data from each device.	
When I open the devices and begin to receive the data. At first, all of the threads were working well. 
After two minutes(or shorter), one or two threads were blocked(the last one still worked well), and the function waitForNewFrame was never return.

I'm trying to find a way to solve this problem.
So I refer to the freenect2 project code.
The problem is that the code "processor_->process(packet)" in function "void DepthPacketStreamParser::onDataReceived(...)"  will never be executed.
Because variables "footer->sequence" and  "footer->subsequence"  are all values greater than 50 thousand...

Is there anyone who can help me?