OpenKinect/libfreenect2 #691: Record the RGB and IR image from the libfreenect2

cforntu, 2016-07-20:
Hello, now we can see the color and ir image from the protonect source code. Could anyone tell me how to modify the code to save the color and IR image?

Additional Information:

robozo, 2016-07-21:
if someone has done this, it would save me some time, as this is this weekend's project.
the problem is, it's a lot of data.  i plan to save the point cloud, plus color data, which is around 45 MB/sec at 30 fps.  might be able to save to SSD disk at that rate, more likely save to memory.
saving the uncompressed RGB is too huge, more like 250 MB/sec.
might be able to save the raw jpeg, but then would have to dig into the "listener" code, rather than just modifying protonect.

djivani, 2016-07-21:
You can use OpenCV to do this. From what I have tried, saving raw RGB is the best bet if you want to maintain the frame-rate, especially if you are working with multiple kinects. At 15fps and 7-8MB/image a SSD should work fine. A HDD might be okay too for just one kinect.:
`cv::Mat(rgb->height, rgb->width, CV_8UC4, rgb->data).copyTo(rgbMat);`
`ss << "Path/image"<<framecount<<".bmp";`
`filename = ss.str();`
`cv::imwrite(filename.c_str(), rgbMat);`

robozo, 2016-07-23:
this issue is a duplicate of #438, #551 and probably can be closed.
the newest version of libfreenect2 apparently has some data logging functions,
but i want a custom data format, so am reinventing the wheel (again).

xlz, 2016-08-09:
Recording with OpenCV, or ROS, or handrolled compression libraries. The big issue is it's just too big for something that is included by default to be useful for everyone. I have plans for such a tool but it's too big to make it useful so far.

robozo, 2016-08-09:
just a note, writing out the point cloud to a SSD was not a problem, on a 2.9 GHz mac laptop
running 10.8.5 even with a browser, etc. open.

while(!protonect_shutdown)
  {
  ...
  write(fd,Mydepth,4*512*424);
  write(fd,Mycolor,3*512*424);
  ...
  }

ran fine at 30 fps, haven't tested what limits are.  SSD's are fast!  they just fill up fast...

xlz, 2016-12-05:
Merge into #438.

closed 2016-12-05 by xlz