OpenKinect/libfreenect2 #693: Depth scale issue in point cloud using OpenNI2 and PCL

dletozeun, 2016-07-22:
Overview Description:

Hello,

I am using PCL (v 1.8.0) OpenNI2 grabber to directly visualize a point cloud extracted from a Kinect v2 using libfreenect2 driver. Depth/color registration is correct but I am noticing that the point cloud is somehow scaled along the sensor z axis. 

[image]

The point cloud is seen from the top and the wooden stool in the middle is a perfect square in reality with right angles.

Is this something you reproduce on your end or am I missing something? I am using PCL OpenNI2Grabber the most basic way possible, extract point clouds as PointXYZRGBA type and use PCLVisualizer class to directly visualize the point cloud without any post processing.

Would attaching the project code help to reproduce the issue?

Thanks

Version, Platform, and Hardware Bug Found:

Windows 10
libfreenect2: master
OpenNI2: kinect2 branch
PCL 1.8.0
VS 2013

Reproducibility:

100% Tested with several Kinect V2 devices.

dletozeun, 2016-08-01:
Found out the issue, camera parameters were not passed correctly to PCL.

closed 2016-08-01 by dletozeun

kaygudo, 2016-08-09:
Hey! can you tell me how to generate point clouds using libfreenect2 or pylibfreenect2 ? possibly by using python?