OpenKinect/libfreenect2 #694: Libusb not found on install build

edward876, 2016-07-22:
I keep getting the following error on install after I enter the line:

`cmake .. -G "Visual Studio 14 2015 Win64"`

```
CMake Error at C:/Program Files/CMake/share/cmake-3.6/Modules/FindPkgConfig.cmake:424 (message):
  A required package was not found
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.6/Modules/FindPkgConfig.cmake:597 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
  CMakeLists.txt:93 (FIND_PACKAGE)`
```

I used the VS2015 command:

`./install_libusb_vs2015.cmd`

To build the the libusb and the libusb_src folders in the depends folder. I am new to cmake so I could be looking over something really simple but I am stumped. Thanks for any help.

chengzijiazu, 2016-07-23:
what is your operating system? I try so many time on win10+VS2015,but it can not perfect! You have better base on win8+vs2015orvs2013.

edward876, 2016-07-25:
I am trying windows 10 and vs2015. Is it really impossible?

dongxugao, 2016-09-01:
I used brew install libusb on mac and still have this problem,

-- using tinythread as threading library
-- Checking for module 'libusb-1.0'
--   Package 'libusb-1.0' not found
CMake Error at /opt/local/share/cmake-3.4/Modules/FindPkgConfig.cmake:360 (message):
  A required package was not found
Call Stack (most recent call first):
  /opt/local/share/cmake-3.4/Modules/FindPkgConfig.cmake:522 (_pkg_check_modules_internal)
  cmake_modules/FindLibUSB.cmake:21 (PKG_CHECK_MODULES)
  CMakeLists.txt:93 (FIND_PACKAGE)....

but I have installed the libusb and got the warning:  Warning: libusb-1.0.20 already installed

xlz, 2016-10-09:
Use some prebuilt binary from https://github.com/OpenKinect/libfreenect2/releases/tag/v0.1.0

Also check the build log of install_libusb_vs201?.cmd to look for any errors that disable the libusbK backend.

xlz, 2016-10-31:
I cannot reproduce this issue.

closed 2016-10-31 by xlz

fliessen, 2019-12-12:
got the same problem with copy the compiled file