OpenKinect/libfreenect2 #695: [Question] Is it possible to visualize raw data using OpenGL ?

moonsh, 2016-07-23:
Hello . 

I would like to visualize raw data which is got from Libfreenect2 using OpenGL.

Therefore, I was trying to include Freeglut but it was failed. 

My goal is visualizing raw data x,y and z as vertexes at OpenGl screen in real time.

Thank you.

robozo, 2016-07-24:
probably a lot easier to use glfw rather than glut.
would advise getting Protonect.cpp example to compile, then look at viewer.cpp which uses glfw.
replace this with your functions using glDrawArrays(GL_POINTS, ...) or whatever.  works great.

the problem with glut is it wants to run an event loop, so do the libfreenect2 "listeners".
good luck!

moonsh, 2016-07-25:
Thanks for advice.  I will try .

closed 2016-07-25 by moonsh