OpenKinect/libfreenect2 #696: Kinect orientation

sdenenberg, 2016-07-25:
When pulling in the data from kinect using libfreenect2, what is the orientation of the kinect sensor by default?  I have a registered depth and rgb image streaming in PCLvisizualizer and am at the point of trying to rectify this data with other sensor inputs.  I'm assuming that the sensor location is (0,0,0) by default, is this correct?  Knowing the orientation of the sensor is necessary as well and I can't find any documentation on this.  

In addition is there any way to set these parameters so that a transformation is done at a lower level?

Thanks,

Scott

jjoachim, 2016-08-01:
The kinect's source is definitely (0,0,0), and I believe it is facing and centered around the +z direction. I use PCL to transform the point clouds. There's almost certainly not a way to do it within the driver.

DISCLAIMER: I am not a developer, just a fellow user trying to be helpful.

xlz, 2016-08-09:
There is plan to add mirroring as device parameters.

xlz, 2016-12-05:
Merge into #172.

closed 2016-12-05 by xlz