OpenKinect/libfreenect2 #697: How to get phase images.

edward876, 2016-07-27:
The kinect 2 take in some number of images and performs calculations on them to give us the intensity and depth values for the IR sensor. Is there any way to get those original images before conversion to the single  512x424 frame? Basically what is the rawest form of data for the IR camera I can get using libfreenect2?

christiankerl, 2016-07-28:
you can get those images by adapting one of the DepthPacketProcessors. for example in CpuDepthPacketProcessor in line 463-471 ( https://github.com/OpenKinect/libfreenect2/blob/master/src/cpu_depth_packet_processor.cpp#L463 ) you get the values of these 9 "raw" images.

edward876, 2016-07-29:
Thank you! I was searching everywhere. However I am a Linux novice (usually I program c++ in VS on windows). How would I go about using the cpu_depth_packet_processor.cpp file/where is its header?

xlz, 2016-08-09:
There is a DumpPacketPipeline which gives you the raw data. The "raw" format is even in the frame->format field. You can determine it. You create a dumppacketpipeline and attach it to a device to get raw frames.

christiankerl, 2016-08-09:
well the dumpdepthpacketprocessor doesn't remove the data compression/encoding, so you don't have meaningful images to look at, right?

xlz, 2016-08-09:
But the user wants the "rawest" data, which is this.

xlz, 2016-09-06:
I think this is resolved.

closed 2016-09-06 by xlz

manideepabc, 2016-10-28:
@edward876 can you please share the code to get the phase images?