OpenKinect/libfreenect2 #698: installation

peter73421, 2016-07-29:
I got notebook not having USB3.0. So I have AXAGO ECU-23R SUPERSPEED (https://az-europe.eu/en/axago-expresscard-34-adapter-2x-usb3-0-renesas-p984710/e)
 
Now I did try  ./bin/Protonect

r59400@r59400:~/work/libfreenect2-master/build$ ./bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 000367554947
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9759
[Error] [protocol::UsbControl] failed to set ir interface state! LIBUSB_ERROR_OTHER Other error. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
r59400@r59400:~/work/libfreenect2-master/build$ 

What causing this problem and how to get running Kintect v2 ?
Thank you.

miguelriemoliveira, 2016-07-29:
Hi Peter,

This could be related #679

peter73421, 2016-07-29:
I have Dell Latitude Dell Latitude E6410 (http://www.cnet.com/products/dell-latitude-e6410-14-1-core-i5-520m-windows-7-pro-2-gb-ram-250-gb-hdd-series/specs/) I use ExpressCard/54 slot. Do you think is the same issue as yours?

peter73421, 2016-07-29:
did you find solution? is that work around to get running Kinect v2 even though?

miguelriemoliveira, 2016-07-29:
In your link I can't find your laptop pci express specifications.

Try [this](http://superuser.com/questions/693964/can-i-find-out-if-pci-e-slot-is-1-0-2-0-or-3-0-in-linux-terminal) to find out which pci express you have.

I also have a Dell Latitude, a more recent model I think, so I am afraid your laptop should have the same issue. But you can easily check.

miguelriemoliveira, 2016-07-29:
No I did not find a solution. Note that I am not sure about that. But if its true, I think that there is no solution. Its like asking if you can run the kinect2 on a USB 2.0. You just can't.

I just use another computer.

Anyway, if you have some solution please share it.

peter73421, 2016-07-29:
My laptop:
dmidecode | grep "PCI"
		PCI is supported
	Type: x16 PCI Express x16
	Type: x1 PCI Express
	Type: x1 PCI Express
	Type: x1 PCI Express
	Type: x1 PCI Express

miguelriemoliveira, 2016-07-29:
I think I used the answer from quazgar. The one that uses the command lspci -vv.

Can you post the output of 

```bash
lspci -vv | grep -E 'PCI bridge|LnkCap'
```

peter73421, 2016-07-29:
lspci -vv | grep -E 'PCI bridge|LnkCap'
00:01.0 PCI bridge: Intel Corporation Core Processor PCI Express x16 Root Port (rev 02) (prog-if 00 [Normal decode])
00:1c.0 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 1 (rev 05) (prog-if 00 [Normal decode])
00:1c.1 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 2 (rev 05) (prog-if 00 [Normal decode])
00:1c.2 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 3 (rev 05) (prog-if 00 [Normal decode])
00:1c.3 PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 4 (rev 05) (prog-if 00 [Normal decode])
00:1e.0 PCI bridge: Intel Corporation 82801 Mobile PCI Bridge (rev a5) (prog-if 01 [Subtractive decode])

miguelriemoliveira, 2016-07-29:
Hi Peter,

I am not sure, since your output does not mention speed. My recommendation is that you find out which is the pci express version and whether or not its compatible with USB 3.0. This way of finding that out worked for me, so I tried nothing else. Try to google to find it.

Regards,

Miguel

peter73421, 2016-07-29:
I got another notebook Dell E6440 with USB3.0 but with Windows 7. How about to run Ubuntu from USB (live) and test Kinect V2 on that Ubuntu instead. Would it be again issue? Maybe not due USB-Kinect bandwidth but Ubuntu performance due to running from USB stick? What do you think?

miguelriemoliveira, 2016-07-29:
I am not sure. But I would not use  a computationally demanding process (as the processing of the kinect2 clouds is) on a Ubuntu live system. Why don't you install Ubuntu side by side with windows? Or, even better, why don't you just erase windows and install Ubuntu :-)?

mariusschweizer1, 2016-11-07:
Hi i try to connect the kinectv2 with my linux computer and i got the same message when i run **./Protonect cpu** that seems to be problem with the bandwith...
I'm using Ubuntu 16.04 with intel i5 chipset (PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express; Intel Corporation 82801 Mobile PCI Bridge). I extended the USB3.0 Card at the PCIe slot.

**robotino@robotino:~/libfreenect2/build$ **./bin/Protonect cpu****
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Ironlake Mobile - 1.7.0
[Error] [VaapiRgbPacketProcessorImpl] vaQueryConfigEntrypoints(display, VAProfileJPEGBaseline, entrypoints, &num_entrypoints): the requested VAProfile is not supported
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @3:6 with serial 001983763147
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9807
**[Error] [protocol::UsbControl] failed to set ir interface state! LIBUSB_ERROR_OTHER Other error. Try debugging with environment variable: export LIBUSB_DEBUG=3 .**
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed

When I check the available USB speed with **lsusb -t | grep xhci**  the USB 3.1-1 seems to be convenient.
robotino@robotino:~$ lsusb -t | grep xhci
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, **5000M**
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M

But **dmesg** output says there is not enough bandwith...
[ 7244.676664] usb 3-1: new SuperSpeed USB device number 5 using xhci_hcd
[ 7247.661450] usb 3-1: New USB device found, idVendor=045e, idProduct=02d9
[ 7247.661457] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7247.661461] usb 3-1: Product: NuiSensor Adaptor      
[ 7247.661464] usb 3-1: Manufacturer: Microsoft Corporation  
[ 7247.664460] hub 3-1:1.0: USB hub found
[ 7247.664891] hub 3-1:1.0: 1 port detected
[ 7248.131783] usb 3-1.1: new SuperSpeed USB device number 6 using xhci_hcd
[ 7248.152495] usb 3-1.1: New USB device found, idVendor=045e, idProduct=02c4
[ 7248.152502] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 7248.152507] usb 3-1.1: Product: Xbox NUI Sensor
[ 7248.152510] usb 3-1.1: Manufacturer: Microsoft
[ 7248.152513] usb 3-1.1: SerialNumber: 001983763147
[ 7695.666388] usb 3-1.1: reset SuperSpeed USB device number 6 using xhci_hcd
[ 7696.147240] **usb 3-1.1: Not enough bandwidth for new device state.**
[ 7696.147254] usb 3-1.1: Not enough bandwidth for altsetting 1

Also i checked the PCIe version but there is no info about PCIE version..:
robotino@robotino:~$ sudo dmidecode | grep "PCI"
		PCI is supported
	Internal Reference Designator: J9C1 - PCIE DOCKING CONN
	Type: x16 PCI Express
	Type: x1 PCI Express
	Type: x1 PCI Express
	Type: x1 PCI Express
	Type: x1 PCI Express
	Descriptor 9: PCI parity error
	Descriptor 10: PCI system error

Can some one help me to fix this problem?
What should i do next to find out if it is a hardware problem?

floe, 2016-11-07:
What USB3 controller does your card use? Try `lspci | grep -i xhci`.

mariusschweizer1, 2016-11-09:
Thank you for response!

When i type in **lspci | grep -i xhci** nothing occur...

**robotino@robotino:~$ lspci**
00:00.0 Host bridge: Intel Corporation Core Processor DRAM Controller (rev 18)
00:02.0 VGA compatible controller: **Intel Corporation Core Processor Integrated Graphics** Controller (rev 18)
00:16.0 Communication controller: Intel Corporation 5 Series/3400 Series Chipset HECI Controller (rev 06)
00:16.2 IDE interface: Intel Corporation 5 Series/3400 Series Chipset PT IDER Controller (rev 06)
00:16.3 Serial controller: Intel Corporation 5 Series/3400 Series Chipset KT Controller (rev 06)
00:19.0 Ethernet controller: Intel Corporation 82577LM Gigabit Network Connection (rev 06)
00:1b.0 Audio device: Intel Corporation 5 Series/3400 Series Chipset High Definition Audio (rev 06)
00:1c.0 **PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 1 (rev 06)**
00:1c.4 **PCI bridge: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 5 (rev 06)**
00:1d.0 USB controller: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller (rev 06)
00:1e.0 **PCI bridge: Intel Corporation 82801 Mobile PCI Bridge (rev a6)**
00:1f.0 ISA bridge: Intel Corporation QM57 Chipset LPC Interface Controller (rev 06)
00:1f.2 IDE interface: Intel Corporation 5 Series/3400 Series Chipset 4 port SATA IDE Controller (rev 06)
00:1f.3 SMBus: Intel Corporation 5 Series/3400 Series Chipset SMBus Controller (rev 06)
00:1f.6 Signal processing controller: Intel Corporation 5 Series/3400 Series Chipset Thermal Subsystem (rev 06)
01:00.0 **USB controller: NEC Corporation uPD720200 USB 3.0 Host Controller (rev 04)**
02:00.0 IDE interface: JMicron Technology Corp. JMB368 IDE controller
ff:00.0 Host bridge: Intel Corporation Core Processor QuickPath Architecture Generic Non-core Registers (rev 05)
ff:00.1 Host bridge: Intel Corporation Core Processor QuickPath Architecture System Address Decoder (rev 05)
ff:02.0 Host bridge: Intel Corporation Core Processor QPI Link 0 (rev 05)
ff:02.1 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor QPI Physical 0 (rev 05)
ff:02.2 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor Reserved (rev 05)
ff:02.3 Host bridge: Intel Corporation 1st Generation Core i3/5/7 Processor Reserved (rev 05)

The extended USB3.0 card that i use requires:
- PCI-Express Revision 2.0
- Full single lane (1x)

Product link:
http://www.logilink.de/Produkte_LogiLink/Interface_Karten/Express_Karten/PCI_Express_Karte-2x_USB_30.htm

mariusschweizer1, 2016-11-09:
OK i got new informations from company Festo who produce this robot (Festo Robotino Premium) with linux platform.
The PCIe slots are version 2.0 and limited to x1 (lane). They said the bandwith is to low. I will try it with a other linux computer.

xlz, 2016-12-05:
Looks like discussion is over.

closed 2016-12-05 by xlz

1329844541, 2019-07-23:
I use the usb3.0 ports