OpenKinect/libfreenect2 #700: TegraJPEG not found - Jetson TX1

marketto89, 2016-08-01:
Hi,

I'm writing because I noticed that flashing my NVidia Jetson TX1 with both the JetPack 2.2 and JetPack 2.2.1 cause libfreenect2 not to find the TegraJPEG Library.

I am able to use libfreenect2 using TurboJPEG, but I wonder if it could be possible to make it work with TegraJPEG given the best performance it should have.

I confirm that with the JetPacks 2.1 and below, libfreenect2 works perfectly with TegraJPEG.

Thank you for your help!

marketto89, 2016-08-01:
Ok, I found that the FindTegraJPEG.cmake does not take into account L4T 24.1 

I created #701 which should fix the issue.

xlz, 2016-11-02:
Should be fixed in 07a7424cc2e22989942818fc74f6d940a4fa6b30.

closed 2016-11-02 by xlz