OpenKinect/libfreenect2 #704: Ubuntu Multiple Kinects

Severn-Vergil, 2016-08-08:
Thank you in advance for your patience!
Overview Description:
I've used two Kinect by ROS iai_kinect2 package on my notebook successfully, however for some reasons I reinstalled my system and reconfigured the environment, now I can not make them work.

Current Version, Platform, and Hardware Bug Found:

uname -a
Linux rabot 4.2.0-42-generic #49~14.04.1-Ubuntu SMP Wed Jun 29 20:22:11 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 1: Dev 16, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 18, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 18, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 18, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 18, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
    |__ Port 2: Dev 15, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 19, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 19, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 19, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 19, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 1: Dev 13, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 2: Dev 12, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 4: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 4: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 7: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 14: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 12M

lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor DRAM Controller [8086:0c04] (rev 06)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller [8086:0c01] (rev 06)
00:02.0 VGA compatible controller [0300]: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller [8086:0416] (rev 06)
00:03.0 Audio device [0403]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller [8086:0c0c] (rev 06)
00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 05)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 05)
00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 05)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 [8086:8c14] (rev d5)
00:1c.2 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d5)
00:1c.3 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #4 [8086:8c16] (rev d5)
00:1c.6 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #7 [8086:8c1c] (rev d5)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 05)
00:1f.0 ISA bridge [0601]: Intel Corporation HM87 Express LPC Controller [8086:8c4b] (rev 05)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c03] (rev 05)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 05)
01:00.0 3D controller [0302]: NVIDIA Corporation GK107M [GeForce GT 750M] [10de:0fe4] (rev a1)
07:00.0 Network controller [0280]: Ralink corp. RT3290 Wireless 802.11n 1T/1R PCIe [1814:3290]
07:00.1 Bluetooth [0d11]: Ralink corp. RT3290 Bluetooth [1814:3298]
09:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader [10ec:5227] (rev 01)
0f:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 0c)

Run LIBUSB_DEBUG=3 ./Protonect on another Kinect:
libusb: error [op_get_configuration] device unconfigured
libusb: error [op_set_configuration] failed, error -1 errno 28
[Error] [protocol::UsbControl] failed to set configuration! LIBUSB_ERROR_OTHER Other error. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:19

dmesg:
[dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/406315/dmesg.txt)

Past Version:
linux kernel 3.16, libusb 1.0.20, installed the libfreenect2 in 2015.12.29;
8GB memory, i7-4702MQ

Additional Information:
It seems weird that the hardware could support two Kinects before and dmesg shows not enough bandwidth now. 
Thank you for your advice again!

xlz, 2016-08-09:
Not enough bandwidth. A hardware problem.

I don't have a solution for this.

closed 2016-08-09 by xlz

Severn-Vergil, 2016-08-09:
@xlz Thank's for your attention! I wonder if it is the hardware problem, as I successfully used two Kinects on my platform before and there are two PCIE-x16 according to the dmidecode result, maybe the PCIs are not allocated well and two Kinects are using the same PCI, and I'm trying to fix this. Are there some parts of code related to the PCI in libfreenect2? Thank you.